ARM-software/CMSIS_5 #314: Cortex-R support

apullin, 2018-02-04:
It appears there is no Cortex-R build support. Parts are on the market now. Is CR4 support planned?

JonatanAntoni, 2018-02-06:
Hi @apullin,

currently we do not have support for Cortex-R class. Adding support for it mainly depends on market pull. Do you have requirements you'd like to share? What would you expect CMSIS for Cortex-R delivers?

Thanks,
Jonatan

apullin, 2018-02-08:
I am presently using a CM4 + CMSIS-DSP to do audio processing. The application is for calculating A-weighted SPL/noise, but many more solutions are on the table, e.g. if big IoT providers and Sensory/chip vendors/etc make trigger-word IP available.

Some CR4 parts are coming on the market in the IoT space, and I am considering options that use those cores. For example, Samsung now ships an "Artik" module that has a CR4 core as an intermediate solution before stepping up to a full Cortex-A Linux platform.

JonatanAntoni, 2018-02-09:
Hi @apullin,

Thanks for your insights to your application ideas based on CMSIS, that's very beneficial. 
In other words you would like to see CMSIS-DSP to be available for Cortex-R4? Hence CMSIS-Core needs to be available as well. What about RTOS (RTX5)?

Would you be able and willing to contribute on this area? I.e. as a first step we would need to have a draft for CMSIS-Core(R) and some startup code for Cortex-R4.

Thanks,
Jonatan

apullin, 2018-02-11:
That is beyond my technical abilities for now. It would be fun to explore it, but there's a lot I don't know there. I have been able to use the prebuilt CMSIS-DSP libraries on M4 just by dropping in the .a and .h files, and not having to go into the low levels of dealing with different programming models, exception models, etc.

But I will definitely contribute anything I come up with.

JonatanAntoni, 2018-02-12:
Hi @apullin,

I appreciate your collaboration. Let's keep this requirement on the backlog.

Please let us know about your questions when you start exploration. You'd be fine adopting CMSIS-Core and system startup code for Cortex-R in the first place.

Thanks,
Jonatan

BabaJaguska, 2019-04-23:
CMSIS-NN would come in handy on Cortex-R too.

alfedotov, 2020-06-19:
I had an attemt to upsteram Core-R about year ago with no luck. But I still see demand.
I have created new pull request there https://github.com/ARM-software/CMSIS_5/pull/945
Please support me if you are interested :)

alfedotov, 2020-06-19:
@JonatanAntoni  May be it worth to create new "future" branch (from devel) if you can't accept these changes on "devel" branch. 
Seems to me it would be more beneficial to have such changes in one central repo instead of searching over github