ARM-software/CMSIS_5 #318: Experiencing strange SVC & PSP behaviour

kakukz493, 2018-02-11:
I need some help with the following issue I'm experiencing with STM32F207VET & CMSIS 5.1.0. Sometimes calling osMutexRelease ends up returning the mutex id and not the result supposed to return (osOk, etc.). Checking the stack there are 2 words from osMutexRelease (r4, lr) and 8 words from the exception stacking because of SVC 0 call (r0-r3, r12, etc.). However looking at SVC_Handler in irq_cm3.s I would expect the return values overwriting the stacked ones after the service function call so unstacking will get the function return values in R0-R1, but it is not happening, stacked register values are left unchanged and Im getting back the mutex id as the function return value. Can any1 shred some lights for me please?

JonatanAntoni, 2018-02-12:
Hi @kakukz493,

What type of HAL initialization code do you use?
Strange behavior like this might be caused by improper interrupt priority settings. Please double check if
- SysTick and PendSV have lowest preempt priority assigned, and
- SVC is configured to use one preempt priority higher.

In CMSIS 5.1.0, i.e. RTX5 5.2.0, `osKernelInitialize` configures the correct priorities. Hence changing interrupt priority related settings after `osKernelInitialize` might cause issues, e.g. by modifying priority grouping.

Best,
Jonatan

kakukz493, 2018-02-12:
Hello Jonatan,

CubeMX generated. Heres the current configuration of the IRQ prios (which is probably the default, untouched).

[image]

On default configuration with the prio group being identical the order - according to the processors programming manual STM32F20xxx - depends on the exception number (lower comes 1st). SVCall has 11, PendSV 14 and Systick is 15. That would mean that SVCall could not be interrupted by Systick or PendSV? Im not debating that this is good, just trying to figure out the current behaviour, and why is it happening.

Checked SVC_Initialize that is trying to configure the the prios and ending up with the following register content in SCB->SHP:

[image]

That would mean SVCall prio 14. Systick 0 and PendSV 15.

Also checked in my code HAL prio config is called AFTER osKernelInitialize (which will of course remove the values set by SVC_Initialize).

Regards,
Zoltan

JonatanAntoni, 2018-02-12:
Hi Zoltan,

Please try to call `osKernelInitialize` again right before `osKernelStart`. This should reliably restore the correct priority settings. With the RTX5 5.2.3 (part of CMSIS 5.2.0) [we moved the priority configuration](https://github.com/ARM-software/CMSIS_5/commit/6d4b6991f3e9e22bab2020882060aa9997deb749) to `osKernelStart` to make this more robust.

Using the `4 bit for preemption priority` setting you should assure that
 - SysTick and PendSV are on priority 15, and
 - SVC is on priority 14.
It's crucial that an SVC is not interrupted by SysTick or PendSV. All those apply changes to internal RTX control structures. Hence it must be assured that those never preempt each other. SVC needs a slightly higher preemption priority to assure its handled with preference. Note that the SVC can only be triggered by application software. Thus it can never happen during SysTick or PendSV. So there's no preemption issue in this case.

Best,
Jonatan

kakukz493, 2018-02-12:
Hi Jonatan,

According to SCB->SHP, after SVC_Initialize the registers are not configured as they should be I think. Please see the attached screenshot of the memory.

0xe000ed20 should look like 0xf0f00000?

JonatanAntoni, 2018-02-12:
Hi Zoltan,

I tried to reproduce and can confirm that default priority settings are slightly different in RTX 5.2.0, i.e.
 - PendSV = 15 (7s1)
 - SysTick = 14 (7s0)
 - SVC = 12 (6s0)

Unfortunately this only works correctly, if you have at least one bit for subpriority. You could try to change your Priority Group setting to 3+1 (instead of 4+0).

If you don't mind please try to upgrade to CMSIS 5.2.0. We have fixed some issues regarding priority configuration in RTX5 5.2.3. In the latest release the priority configuration is done during `osKernelStart` and the CubeMX configuration should not harm anymore.

Best,
Jonatan

kakukz493, 2018-02-12:
Hi Jonatan,

Upgrading to CMSIS 5.2.0 after calling SVC_Setup from osKernelStart still getting the same register values as before. I modified the prio group setting to 3+1. I attached the screenshot of the memory address right after SVC_Setup. SysTick still getting prio 0.

[image]

JonatanAntoni, 2018-02-12:
Hi Zoltan,

SysTick priority is configured by `OS_Tick_Setup` called from `svcRtxKernelStart` after `SVC_Setup`. If you check the priorities after `osKernelStart` you should see the correct values, i.e. set a breakpoint in your first thread function.

Best,
Jonatan

kakukz493, 2018-02-12:
Hello Jonatan,

Sorry that was an oversight on my part. I reorganized the code to have the HAL init part before the kernel start. The prio config now looks like this after kernel start:

[image]

SysTick and PendSV both on same prio (15), SVC is on 12. Is this the expected config?

JonatanAntoni, 2018-02-13:
Hi Zoltan,

Assuming you still have Priority Group setting at 3+1 that's correct. Going back to 4+0 should result in the 15,15,14 stated earlier. Does it run without strange issues, now?

Cheers,
Jonatan

kakukz493, 2018-02-13:
Hello Jonatan,

I'm already back on prio group setting 4, the settings are back to 15,15,14. I have not experienced my issue so far. Thank you for your help.

closed 2018-02-13 by kakukz493