ARM-software/CMSIS_5 #319: get tick of timer

juliobc, 2018-02-12:
Is there an API to get the Tick parameter of the virtual timer to know how much time is left?
Regards

JonatanAntoni, 2018-02-12:
Hi @juliobc,

No, we don't have such a function available.
I would like to record this issue as a feature request. Can you please provide slightly more details about how you'd like to use the virtual timer?

Thanks for your contribution,
Jonatan

juliobc, 2018-02-12:
Hi @JonatanAntoni 
I would like to execute a function only when the remaining time of a timer is less than 60 minutes.
Thank you

JonatanAntoni, 2018-02-12:
Hi @juliobc,

Did you close the issue intentionally? Or would you like to see such an API enhancement to become available someday in the future? In the latter case I'd like to keep you request open on the backlog.

Cheers,
Jonatan

juliobc, 2018-02-12:
Hi @JonatanAntoni 
Sorry, I thougth it would be closed. I would like to have this API enhancement.
Regards

JonatanAntoni, 2019-03-12:
@ReinhardKeil, @RobertRostohar: Any thoughts about this one? Its still in review, hence we should decide to accept the request for enhancement or decline it.

RobertRostohar, 2019-03-13:
Having a function to retrieve the timer remaining time before it expires makes sense and would cover certain use cases. We should consider adding such a function.