ARM-software/CMSIS_5 #320: bitwise shift unexpected behavior

jide-huang, 2018-02-13:
in following two function:
arm_softmax_q7.c
arm_softmax_q15.c

line 95:
sum += 0x1 << (vec_in[i] - min);

and line 101:
p_out[i] = (q15_t) __SSAT(((0x1 << (vec_in[i] - min + 14)) / sum), 16);

left bitwise shift might larger than 32

liangzhen-lai, 2018-02-13:
Hi @jide-huang, 

Thank you for raising the issue. The left-shift overflow issue is handled by adjusting the value of _min_ to be close to _max_, so that the range of _vec_in[i] - min_ is constrained. You can refer to, for example, line 82 to 89 in arm_softmax_q7.c for more details.

Thanks,
Liangzhen

jide-huang, 2018-02-14:
Hi

But there still another question:

The result of:
vec_in[i] - min
Will became negative, and small than -32.
In this case, we still get an undefined behavous.

I  suggest limiting the result of "vec_in[i] - min" to value between -31~31 before bitwise shifting.

liangzhen-lai, 2018-02-14:
Hi @jide-huang,

Yes, the value of _vec_in[i] - min_ can be negative. According to the Armv7-M architecture reference manual, the left shift shift instruction will cast the value into uint8 and shift the value accordingly. If it is, for example -33, it is effectively doing (<<223). But you are right, there may be issue when the value is, for example -255, it will do (<<1). 

We will look into this one and apply the appropriate fix.

Thanks,
Liangzhen

liangzhen-lai, 2018-02-22:
#324

JonatanAntoni, 2018-02-23:
Hi @jide-huang,

May I ask you to double check the provided fix, please? Does this change solve your issue? Any concerns left?

Thanks,
Jonatan

JonatanAntoni, 2018-06-27:
Closing this issue due to inactivity. Please feel free to reopen if the problem still exists.

closed 2018-06-27 by JonatanAntoni