ARM-software/CMSIS_5 #321: POSIX pthread

edbek, 2018-02-19:
When there is a layer of POSIX pthread ?

JonatanAntoni, 2018-02-19:
Hi @edbek,

thanks for getting in touch. I guess you are referring to CMSIS-RTOS2 API and RTX5 implementation of it.

We have no plans to enhance the APIs towards POSIX compatibility. As a future enhancement we have C++ standard thread library support on our backlog. As of today I cannot tell you details about the schedule.

May I ask you which functionality you are missing from CMSIS-RTOS2 API? What aspects of POSIX do you think would be great to have?

Best,
Jonatan

edbek, 2018-02-23:
I mean POSIX support for FreeRTOS as in 
http://dev.ti.com/tirex/#/?link=Software%2FSimpleLink%20CC3220%20SDK%2FSimpleLink%20Academy%2FLabs%2FFreeRTOS%2FFreeRTOS%20Basics

JonatanAntoni, 2018-02-23:
Hi @edbek,

The TI subset of the POSIX functions for RTOS seems to be comparable to CMSIS-RTOS2 API on the first glance. It should be possible to map these POSIX functions to CMSIS-RTOS2 ones. 

What would be the benefit for your application to have the POSIX API instead of the CMSIS-RTOS2 API?

Any valuable contribution is appreciated. 

Best,
Jonatan

edbek, 2018-02-23:
easy transfer of ready-made POSIX software to the FreeRTOS platform

ilg-ul, 2018-02-23:
> ready-made POSIX software

This is a nice feature, and makes great demos, but for real applications it is not very realistic.

Embedded applications may also need to control details like memory allocation for the treads, queues, etc, and this is not very well supported (or, if supported, is rarely used) for POSIX applications.

JonatanAntoni, 2018-02-26:
Hi all,

@ilg-ul that's actually why we never tried to be POSIX compliant, so far. But even if its only for demo or fast proof-of-concept work it might be beneficial.

To be honest I am not aware of any ready-made POSIX software that we might take a closer look into regarding the API requirements. I guess you have some stuff in mind, @edbek? Please feel free to move along. It would be great to see what could be achieved with a POSIX-enhancement to CMSIS-RTOS2 API.

Cheers,
Jonatan

ilg-ul, 2018-02-26:
FYI, in µOS++ I already have a standard ISO/ANSI C++ threads implementation, and it is fully functional.

It is quite impressive to copy/paste examples from the books and be able to run them directly, including creating threads with variable number of arguments.

However, as I already said, for real application, people use the native C++ APIs, which provide a better control on where different resources are allocated.

In the next iteration of µOS++ I'll probably add the standard POSIX threads C API too, but it'll be be mostly for completeness, because I don't expect it to be widely used.

edbek, 2018-02-26:
Well, for example, there is a project called pjsip (https://github.com/pjsip/pjproject), in which the POSIX is used, and the transfer of this project to the FreeRTOS would be extremely interesting..

JonatanAntoni, 2018-02-26:
Sounds interesting. I expect this one to have a dependency on an IP stack? So one would need more than only RTOS APIs. I guess it can be a good show case to have POSIX wrappers implemented for CMSIS-RTOS2 (RTX5, or FreeRTOS natively) and lwIP (or MDK-TCP/IP). If you can come up with an implementation any contribution is appreciated.

edbek, 2018-02-26:
Yes, the project simply adds the lwIP. It would be good to get the opinion of Liviu Ionescu (ilg-ul).