ARM-software/CMSIS_5 #322: i.MX7 M4 Atomic Cache Bug

ablaylock, 2018-02-19:
The i.MX7 M4 has an atomic cache bug detailed here:

https://rschaefertech.wordpress.com/2018/02/17/imx7-hardware-bug/

In this writeup on the bug they work around the bug by placing all atomic variables in the TCM via a linker script for FreeRTOS. Is there a similar set of symbols implementers can target in their linker script to place all atomics for RTX in TCM?

JonatanAntoni, 2018-02-20:
Hi @ablaylock,

Using RTX5 you have two options. Please be aware that atomic variables are widely spread over the control blocks. Thus we cannot easily put only a few atomics to a specific region.

You need to place all the control blocks to a non-cacheable memory. This can be achieved by forcing placement of sections `bss.os*` and `.data.os*`, see [rtx_lib.c](https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/RTOS2/RTX/Source/rtx_lib.c).

Another option would be to disable the usage of `LDREX` and `STREX`, completely. This can be achieved using RTX5 source code (not pre-built library) and setting pre-processor define `EXCLUSIVE_ACCESS=1`. Instead of using exclusive access instructions the interrupts are disabled during critical regions.

Please let me know if one of the above solutions fits your needs.

Best,
Jonatan

ablaylock, 2018-02-20:
Jonatan,

I think one of the solutions will be sufficient to solve the problem. Thank you for your prompt answer.

closed 2018-02-20 by ablaylock