ARM-software/CMSIS_5 #325: NN bug in example code?

jafermarq, 2018-02-22:
I'm compiling and running the NN CIFAR-10 example code on a NUCLEO_F767ZI using the terminal on MacOS. The NN execution pipeline (described in `arm_nnexamples_cifar10.cpp`) seem to be working until it reaches the fully connected layer. There it gets stuck. I found that the reason for this is the line: `arm_q7_to_q15_reordered_no_shift(pV, vec_buffer, dim_vec)`, found at the top of `arm_fully_connected_q7.c`. When, uncommented, the entire NN gets executed but I feel like the output is wrong: 
    0:0
    0:0
    0:0
    0:0
    0:0
    0:0

liangzhen-lai, 2018-02-22:
Hi @jafermarq ,

The function `arm_q7_to_q15_reordered_no_shift(pV, vec_buffer, dim_vec)` is implemented [here](https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/NN/Source/NNSupportFunctions/arm_q7_to_q15_reordered_no_shift.c). Can you check where it gets stuck? This function is necessary for the function to operate correctly. 

The printf output is` image_label_id: confidence`. The confidence value generated by softmax function is of format Q7 (i.e., 128 -> 100% confidence). It may not always add up to exactly 128 due to rounding errors. Below is the expected output for your reference:

> start execution
> 0: 0
> 1: 0
> 2: 120
> 3: 3
> 4: 3
> 5: 0
> 6: 0
> 7: 0
> 8: 0
> 9: 0
> 

Thanks,
Liangzhen

jafermarq, 2018-02-22:
Hi @liangzhen-lai ,

Thank you for the clarification on the output's meaning. I narrowed down a bit more where the problem is. Inside `arm_q7_to_q15_reordered_no_shift(pV, vec_buffer, dim_vec)`, the execution enters the first while loop inside `#ifndef ARM_MATH_CM0_FAMILY`, where `blkCnt = 128` but the end of the while loop is not reached. Not even once. I'll find the exact line cause the problem later today. Hope this info helps.

jafermarq, 2018-02-22:
UPDATE. It turns out that by just adding a few `printf` statements, the code works.... Concretely, this is how I made the code work by adding two `printf` calls in the while loop:

     printf("%d\n",blkCnt); \\ <---- ADDED
     while (blkCnt > 0u)
     {
        /* C = (q15_t) A << 8 */
        /* convert from q7 to q15 and then store the results in the destination buffer */
        in = *__SIMD32(pIn)++;
        /* rotatate in by 8 and extend two q7_t values to q15_t values */
        in1 = __SXTB16(__ROR(in, 8));

        /* extend remainig two q7_t values to q15_t values */
        in2 = __SXTB16(in);
        printf(".");  \\ <---- ADDED
    #ifndef ARM_MATH_BIG_ENDIAN
        *__SIMD32(pDst)++ = in2;
        *__SIMD32(pDst)++ = in1;
    #else
        *__SIMD32(pDst)++ = in1;
        *__SIMD32(pDst)++ = in2;
    #endif

        /* Decrement the loop counter */
        blkCnt--;
        printf(".");  \\ <---- ADDED
    }

Keeping the code as above will make the program run properly and outputting the right result (as shown in @liangzhen-lai 's comment). Removing any of those `printf` will cause the execution of the program to get stuck right after entering the while loop (when called from `arm_fully_connected_q7.c`) as described in my first comment. As pointed in [this](https://stackoverflow.com/questions/29313067/why-does-inserting-a-printf-statement-make-my-function-work-correctly) thread, this could be due to an invalid operation...

liangzhen-lai, 2018-03-01:
Hi @jafermarq ,

Thank you for getting this detailed information. What compiler setup did you use for this one?

Liangzhen

jafermarq, 2018-03-01:
I'm using `mbed-cli (1.3.0)` on macOS and I've configured it to use the `v7-2017-q4-major` toolchain. I did so by running: `mbed config -G GCC_ARM_PATH <pathToToolChaing>/gcc-arm-none-eabi-7-2017-q4-major/bin`.

jafermarq, 2018-03-03:
I've noticed I get some warnings above some of the lines where I need to place the `printf()` in order to make my code work. The warning is `dereferencing type-punned pointer will break strict-aliasing rule` and appear on the lines in the while loop using `*__SIMD32()`, which [according to this StackOverflow post](https://stackoverflow.com/questions/8824622/fix-for-dereferencing-type-punned-pointer-will-break-strict-aliasing), it can lead to undefined behavior. 

I've tested my code also on Ubuntu 16 and on a NUCLEO_F746ZG board and I experience the same effect.

iabdalkader, 2018-03-04:
@liangzhen-lai Hi, I'm getting the expected printf output running the example on an F7 board using the built-in image. But when I try to run the network on different images th classification is always wrong. Can you please tell what is the expected input image format ? I convert the image to an array of RRR..GGG..BBB is this correct ?

liangzhen-lai, 2018-03-04:
@iabdalkader, The image should be Height-Width-Channel, i.e., RGB of the first pixel, followed by RGB of the second pixel, etc. The network is trained using the setup in Caffe where input data is raw RGB image minus data set mean image (i.e., mean.binaryproto).

iabdalkader, 2018-03-04:
@liangzhen-lai Hi, thanks for the info.

> minus data set mean image (i.e., mean.binaryproto).

Do you mean I should subtract the mean of the R, G, B channels first ?

iabdalkader, 2018-03-04:
@liangzhen-lai Okay I see what you mean now, I converted the mean.binaryproto from the caffe example to c header, subtracted from the input image but the results are no better. I know it doesn't work is not really an issue I can report,  wish I could provide more specific info. Thanks anyway for the help.

kwagyeman, 2018-03-04:
Maybe a code for feeding any 32x32 image to the network would be good?

iabdalkader, 2018-03-04:
@liangzhen-lai Seems to be working fine (81% is that good enough ?) if I use the arm_fully_connected_q7_opt function. If I used arm_fully_connected_q7 almost all of the images are wrong.

@kwagyeman @liangzhen-lai would be nice to convert mean image to C header and include that with the library. I wasted a lot of time trying to build caffe just to load the binary mean image.

liangzhen-lai, 2018-03-05:
@iabdalkader : If you generate image data from caffe directly, you don't need this as caffe will take care of the mean.binaryproto subtraction. 81% matches what we got when training the network in Caffe. You should use `arm_fully_connected_q7_opt` function as the weight formats are different for `arm_fully_connected_q7_opt` and `arm_fully_connected_q7`.

@kwagyeman : good suggestion. We will maybe include a example python script to generate the image input data from the dataset.

For your reference:
Below is some snippet of the code we used to generate the image data. You can refer to this as an example of how to generate HWC image data. We will include maybe a python script to 

```
net = caffe.Net('cifar10_full.prototxt', 'cifar10_full_iter_80000.caffemodel.h5', caffe.TEST)

# N C H W

batch_size = net.blobs['data'].data.shape[0]
image_channel = net.blobs['data'].data.shape[1]
image_size_y = net.blobs['data'].data.shape[2]
image_size_x = net.blobs['data'].data.shape[3]

for i in range(10):
  out = net.forward()
  for n_img in range(batch_size):
    image = np.zeros(image_channel*image_size_y*image_size_x, dtype=np.int8)
    for i_ch in range(image_channel):
      for i_y in range(image_size_y):
        for i_x in range(image_size_x):
          image[(i_y*image_size_x+i_x)*image_channel+i_ch] = net.blobs['data'].data[n_img][i_ch][i_y][i_x]
```

iabdalkader, 2018-03-05:
@liangzhen-lai Note we're running this code on images in real time on a camera.

jafermarq, 2018-03-05:
@liangzhen-lai , I've created a simple repo hoping it will be easier for you (or @iabdalkader) to spot my error. The code there is exactly the same as in the CIFAR-10 NN example. I've pretty much explained the same as here: (1) that my example program gets stuck when executing the fully connected layer and (2) that I "solve" that issue by adding `printf()` statementes after certain data accessing instructions. 

Could you clone that repo and test if you encounter the same issue? @iabdalkader, I'd really appreciate if you have also a chance to try it. Thank you

iabdalkader, 2018-03-05:
@jafermarq Hi, it looks okay, I can't test it I don't have a NUCLEO_F767ZI board and don't know how to use the mbed tools. One thing I noticed, your printf shouldn't have any effect the code should be compiled for little endian target ? Anyway you should try the following Makefile CFLAGS:
```
-D__FPU_PRESENT=1 -D__VFP_FP__  -DSTM32F765xx -D$(CFLAGS_MCU) -DARM_MATH_CM7
 -fsingle-precision-constant -Wdouble-promotion -mcpu=cortex-m7 -mtune=cortex-m7
 -mfpu=fpv5-sp-d16 -mfloat-abi=hard
```

liangzhen-lai, 2018-03-05:
@jafermarq , Thank you for creating the repo. I can reproduce the results that program stuck at the same exact location you were reporting. My guess is that this may be an instruction alignment issue but not 100% sure. 
I have tried a simple fix here that use `-O3` gcc flag instead of `-Os`. All you need to do is to create a .json file (e.g., [here](https://github.com/ARM-software/ML-KWS-for-MCU/blob/master/Deployment/release_O3.json
)) and add `--profile target_O3.json` to your mbed compile command.

Also, as suggested by @iabdalkader, you should use the function `arm_fully_connected_q7_opt`. 

Let me know if this fixes the issue.

Thanks,
Liangzhen

jafermarq, 2018-03-05:
@liangzhen-lai , Amazing! the `.json` file solves the issue. Thank you!!

Baaaaaarry, 2018-05-23:
I directly used the sample code of cifar-10 project in Keil uVision. It runs no error.

JonatanAntoni, 2018-05-25:
@jafermarq, please close this issue if your request has been solved. Thanks.

closed 2018-05-25 by jafermarq