ARM-software/CMSIS_5 #326: About MDK file system component- nand flash transaction layer

salfe, 2018-02-26:
Recently when I was trying to enable the MDK NFTL, I found one question about MDK NFTL library:

Current CMSIS nand driver need to provide below interface,
ARM_DRIVER_NAND Driver_NAND_(NAND_DRIVER) = {
  GetVersion,
  GetCapabilities,
  Initialize,
  Uninitialize,
  PowerControl,
  DevicePower,
  WriteProtect,
  ChipEnable,
  GetDeviceBusy,
 SendCommand,
  SendAddress,
  ReadData,
  WriteData,
  ExecuteSequence,
  AbortSequence,
  Control,
  GetStatus,
  InquireECC
};
The MDK NFTL library only use the SendCommand/ SendAddress/ ReadData/ WriteData per my debug, if a host controller doesn’t support send command/address/data separately, then software need cost effort on cache the command with address.
And most of the host controller support execute a command sequence only(ExecuteSequence). 

why does MDK-NFTL not call ExecuteSequence ?  Does this is a limitation of the MDK-FTL? Or did I missed some configurations?

Thanks!

jkrech, 2018-02-26:
Hi,
please note that this is the CMSIS repository and not the MDK-Middleware support portal.
For MDK-Middleware product inquiries and support requests, please go to:
https://developer.arm.com/support
Thanks a lot.

Kind regards,

Joachim Krech

salfe, 2018-02-27:
Ok, so can i discuss cmsis_nand_flash driver here?

JonatanAntoni, 2018-03-02:
Hi @salfe,

yes, if you have questions or concerns about CMSIS components (that are part of this repo) please feel free to post them.

Cheers,
Jonatan

salfe, 2018-03-07:
Hi @JonatanAntoni 
I want to discuss the cmsis nand flash driver interface,
currently cmsis nand driver require user provide send command/send address/write data/read data, as i know, most of the nand command combine with data or address and most of the host controller don't support send command/adress/data separately.
I think send address  api is not reasonable, it is not easy for user to provide, i have enabled the cmsis nand driver, to implement this api, i have to cache the command which has address and send it together with address, this is a ugly workaround,

JonatanAntoni, 2018-03-07:
Hi @salfe,

the current interface is based on some market analysis we did a couple of years ago.

I guess you are working on another NAND driver? Can you let us know which one you are referring to? If you state
> [..] most of the host controller don't support send command/address/data separately.
which controllers do you have in mind? 

If you could provide some more details that would help our engineering team to analyze nowadays requirements. We will provide more feedback as soon as we sorted out the details.

Thanks,
Jonatan

salfe, 2018-03-08:
Hi @JonatanAntoni 
Yes, i am working on a new nand driver.I was trying to use cmsis nand driver to enable MDK file system FTL component to do some tests.
I am using NXP's i.mx6RT semc host controller, this ip doesn't not support send command and address separately,
From my point, even if a ip support send command/address separately, such cmsis nand flash interface will reduce the performance.
It is ok if application use excute sequence api instead of send command /address.
Personally i think combine send command/address to a single api could improve the compatibly and performance of the cmsis nand driver, such as: SendCommand(cmd, address).

JonatanAntoni, 2018-03-08:
Hi @salfe,

thanks for pointing us to the device you are using. Having a CMSIS-Driver for i.MX RT would be beneficial I think. Any contribution you are willing to share is appreciated. We will provide more details when we had the time to check the details.

Cheers,
Jonatan

ReinhardKeil, 2018-05-30:
Note: these drivers are in the works and should appear shortly

ReinhardKeil, 2018-06-21:
The related pack with CMSIS-Driver "Keil.iMXRT105x_MWP.1.0.0.pack" is now available, therefore I'm closing this issue.

closed 2018-06-21 by ReinhardKeil