ARM-software/CMSIS_5 #327: How are the `bias_shift` and `out_shift` defined in `arm_fully_connected_q7`

maelp, 2018-02-26:
It is not completely clear from the code how the `bias_shift` and `out_shift` are computed for each quantized layer for https://github.com/ARM-software/CMSIS_5/blob/f8c5612872b2c021de2a011b63b4f51f613f985e/CMSIS/NN/Source/FullyConnectedFunctions/arm_fully_connected_q7.c#L68

liangzhen-lai, 2018-03-01:
Hi @maelp ,

The data are assumed in dynamic fixed-point format. For example, the input of q7_t type can be Q4.3 where the actual represented value is the int8 value divided by 2^3, e.g., 0x13 -> 19/8, 0xF1 -> -15/8. 

If you start with a floating-point based network, the quantization process should give the data format for input, weights, bias and output. The `bias_shift` and `out_shift` are used to adjust the difference between different quantized values. For example, if the input is Q5.2, weight is Q1.6, bias is Q4.3 and output is Q3.4, the product of input and weight is Q23.8 with a 32-bit accumulator. In this case, `bias_shift` is 5 (i.e. left shift Q4.3 bias by 5 bits to match Q23.8) and `out_shift` is 4 (i.e., right shift Q23.8 by 4 to match Q3.4 output). 

You can also refer to this for more details:
https://github.com/ARM-software/ML-KWS-for-MCU/blob/master/Deployment/Quant_guide.md
We will add more explanation into the documentation. Let me know if you have more questions.

Thanks,
Liangzhen

maelp, 2018-03-01:
Thanks! Would you also recommend to do negative shift if it is needed, or in this case keep the most precise?

For instance if the values were those

input Q5.2
weight Q7.1
bias Q3.4
output Q3.4

input * weight Q28.3
bias_shift -1
out_shift -1

liangzhen-lai, 2018-03-01:
Hi @maelp ,

Based on our own experience and use cases, output will need to be right-shifted if the input/weight and output have the same bit-width (e.g., q7_t or q15_t). Bias right-shift may not be necessary as this will lose the LSBs and can be done off-line as well.

If you see your quantized networks or use cases that will require either bias right-shift or output left-shift, we can add support to that. Since the code is open-sourced, it should also be pretty straightforward for you to create your own version.

Also, we are in the process of updating the code to use only unsigned int types for shift operations. You can refer to #320 and #324 for more details.

Let me know if this answers your question.

Thanks,
Liangzhen

baileo, 2018-05-04:
Why you assume that the input data is Q5.2 ?
How do you define your output data formate is Q3.4 ?

Tryljs, 2018-06-28:
Why you assume that the input data is Q5.2 ?
How do you define your output data formate is Q3.4 ?

Tryljs, 2018-06-28:
Dear @maelp, @liangzhen-lai :
I had train a mnist mode with tensorflow, and I had used cmsis-nn to run it with floating-point. The accuracy of the model is 0.965 with floating-point, but when I used Q7 to run the model and it had only 0.105 accuracy. I modified the values of bias_shift and out_shift , but the result were not ideal.
So, how to get the values of bias_shift and out_shift ? Can you give me a demo?

kwagyeman, 2018-06-28:
Hi, we only managed to get this working after ARM released their conversion script:

https://github.com/ARM-software/ML-examples/tree/master/cmsisnn-cifar10

If you want to be able to dynamically load a network we've been building out this feature for M4/M7 processors with the OpenMV Cam.

loushu12138, 2019-06-25:
> Why you assume that the input data is Q5.2 ?
> How do you define your output data formate is Q3.4 ?

I have the question too!

llefaucheur, 2019-06-25:
We wrote a tutorial on quantization topics [[here](https://developer.arm.com/solutions/machine-learning-on-arm/developer-material/how-to-guides/converting-a-neural-network-for-arm-cortex-m-with-cmsis-nn/single-page)]
The rule of thumb is to have a look at the "gaussian" curves of your unquantized data and avoid having more than few percents outside of the quantization range.

closed 2021-01-07 by felix-johnny