ARM-software/CMSIS_5 #331: [CM7][dcache] Issue on enabling dcache when it's been enabled already under certain circumstances

jhqian, 2018-03-07:
In core_cm7.h, SCB_EnableDCache, it will invalidate all the sets/ways then enable it no matter if d$ is enabled or not. It might cause issue when
- dcache is enabled
- stack is located in cacheable memory

Because compiler might not store the local variable "sets" and "ways" in to registers.

JonatanAntoni, 2018-03-08:
Hi Ryan,

you are right, invalidating the cache while it is enabled and used can cause serious issues.

The core functions provided are very low level. So I would expect one to use the SCB_EnableDCache function only if the DCache was not enabled before. I admit that might be arbitrarily wrong.

We could add a condition to actually skip the function if the cache is already enabled. The downside is additional overhead I would like to prevent. Or is this just a matter of documentation?`

Please let me know about your opinion. Which implementation would you expect to be provided by CMSIS?

Thanks,
Jonatan

jhqian, 2018-03-12:
Hi Jonatan,

how about using the keyword of "register" for "sets" and "ways"? although it is just a hint for compiler. if it's not an option, I'd like to have the check before enabling d$. documentation would be my last choice.

Thanks.
Regards,
Ryan

GuentherMartin, 2018-10-12:
Hello @jhqian could you please check the changes and close the issue.

closed 2018-10-12 by jhqian