ARM-software/CMSIS_5 #333: Enumeration Values having same value not allowed in SVD files

radhika-raghavendran, 2018-03-08:
Hi,

I would like to modify SVD file for my microcontroller to include duplicate values in the Enumeration Value field. What I would ideally like to have is something like this when I generate a header:

`typedef enum {                                  /*!< XBAR_SPI_CFG   */
  XBAR_IN_SPI_MISO_NONE        = 0,     
  XBAR_IN_SPI_MISO_PA2           = 1,     
  XBAR_IN_SPI_MISO_PB2           = 2,     
  XBAR_IN_SPI_MISO_PB5           = 3,     
  XBAR_IN_SPI_MISO_PC1           = 4,     
  XBAR_IN_SPI_MOSI_NONE       = 0,
  XBAR_IN_SPI_MOSI_PA2           = 1,   
} XBAR_IN_SPI_CFG_IN_Enum;`

I made modifications to the SVD file, added the enumerationValue field with a name and same value as the one used previously, but it does not show up when I generate a header. Is there any way I can modify the SVD to generate a header like the one shown above?

Thank you,
Regards,
Radhika

JonatanAntoni, 2018-03-08:
Hi Radhika,

I guess SVDConv creates the warning M333 and ignores the second enumeratedValue from your SVD file. This seems to be a duplicate to #issue 198.

May I ask you to give us a rational about what you want to achieve with having multiple names for the same value? Please put your comments to issue #198 so that we have the discussion in a single place.

Please close this issue if you agree that it is a duplicate.

Thanks,
Jonatan

radhika-raghavendran, 2018-03-09:
Discussion to continue in Issue #198

closed 2018-03-09 by radhika-raghavendran