ARM-software/CMSIS_5 #334: ICSR defines differ in name between v6m/v7m cores and v8m

galak, 2018-03-09:
In files like CMSIS/Core/Include/core_cm7.h we have:

```
/* SCB Interrupt Control State Register Definitions */
#define SCB_ICSR_NMIPENDSET_Pos            31U                                            /*!< SCB ICSR: NMIPENDSET Position */
#define SCB_ICSR_NMIPENDSET_Msk            (1UL << SCB_ICSR_NMIPENDSET_Pos)               /*!< SCB ICSR: NMIPENDSET Mask */
  
#define SCB_ICSR_PENDSVSET_Pos             28U                                            /*!< SCB ICSR: PENDSVSET Position */
#define SCB_ICSR_PENDSVSET_Msk             (1UL << SCB_ICSR_PENDSVSET_Pos)                /*!< SCB ICSR: PENDSVSET Mask */
```

however in CMSIS/Core/Include/core_cm33.h:
```
/* SCB Interrupt Control State Register Definitions */
#define SCB_ICSR_PENDNMISET_Pos            31U                                            /*!< SCB ICSR: PENDNMISET Position */
#define SCB_ICSR_PENDNMISET_Msk            (1UL << SCB_ICSR_PENDNMISET_Pos)               /*!< SCB ICSR: PENDNMISET Mask */
  
#define SCB_ICSR_PENDNMICLR_Pos            30U                                            /*!< SCB ICSR: PENDNMICLR Position */
#define SCB_ICSR_PENDNMICLR_Msk            (1UL << SCB_ICSR_PENDNMICLR_Pos)               /*!< SCB ICSR: PENDNMICLR Mask */
```

is this on purpose?

galak, 2018-03-10:
I guess this is because the arch spec renamed the bit, however it would be useful from a software perspective to have the old name work on v8m from a compat perspective.

JonatanAntoni, 2018-05-07:
Hi @galak,

I recently added backward compatible definitions. Can you please double check and close?

Thanks,
Jonatan