ARM-software/CMSIS_5 #335: Meaning of complex FFT

escherstair, 2018-03-15:
[Real FFT Functions](https://www.keil.com/pack/doc/CMSIS/DSP/html/group__RealFFT.html) calculate FFT of real data samples.
The output is a complex sequence
`X = { real[0], imag[0], real[1], imag[1], real[2], imag[2] ... real[(N/2)-1], imag[(N/2)-1 }`
than can be used to calculate the magnitude and the phase of the FFT.
As an example, if `real` is greater than zero and `imag` is less than zero, the angle is in the [fourth quadrant](https://amsi.org.au/ESA_Senior_Years/SeniorTopic2/2d/2d_2content_2.html).
Which angle is supposed to return from Real FFT Functions?
If input sequence is
`X[n] = A*cos(2*pi*n/N - a)`
I suppose to get
`a = atan2(imag, real)`

But it seems that I get
`a = -atan2(imag, real)`

Could you clarify this (and add to the documentation the relevant information), please?

llefaucheur, 2018-03-20:
Hi escherstair, when we implement atan2(y,x), the reference output will be the ones from Matlab.

escherstair, 2018-03-20:
Hi @llefaucheur 
I don't understand your explanation.
My question refers to the meaning of real and imag in the output of Real FFT, related to `a` (i.e. the phase of the `cos()` function.
Could you write it with different words, please?

llefaucheur, 2018-03-20:
The reference code is Matlab, without interpreting the relation between the input signal and the phase of the FFT output.  Which angle is supposed to be returned by Matlab's atan2() from your equation ?

escherstair, 2018-04-16:
I thik that the result from Real FFT functions is correct.
If input sequence is
`X[n] = A*cos(2*pi*n/N **+** a)`
`a` can be calculated as
`a = atan2(imag, real)`
where `real` and `imag` are the results from FFT.

llefaucheur, 2018-04-16:
Alright, can we close this issue ?
Kind regards, Laurent

escherstair, 2018-04-16:
Yes.

closed 2018-04-16 by llefaucheur