ARM-software/CMSIS_5 #336: No return from System Reset

hparracho, 2018-03-15:
I think the attribute `__NO_RETURN` could be added to all the `__STATIC_INLINE void __NVIC_SystemReset(void)` function signatures to shut up the compiler warning when compiling with -pedantic turned on:
warning: 'noreturn' function does return.

This happens when a "no return" function calls the __NVIC_SystemReset() hopping it will indeed not return.

Example -  In my code I redefine a newlib's _exit() stub:
```
extern "C" __attribute__ ((weak, noreturn)) 
void _exit(int code __attribute__ ((unused)))
{
#if !defined DEBUG
    NVIC_SystemReset();  
#else  
    while(true)  
    {}  
#endif  
}
```

JonatanAntoni, 2018-03-15:
Hi @hparracho,

Thanks for your feedback. I'll take a look into that. On the first glance it looks viable to me.

Cheers,
Jonatan

hparracho, 2018-04-09:
Hi @JonatanAntoni,

Issue solved. Compiler is happy now and I am too! :-)
Closing.

Thanks,
Helder.

closed 2018-04-09 by hparracho