ARM-software/CMSIS_5 #337: osTimerNew always return 0

Pupil2013, 2018-03-15:
I can't create a timer ,as the osTimerNew always return 0,where is the problem?

JonatanAntoni, 2018-03-16:
Hi @Pupil2013,

please add some more information on what exactly you are trying to achieve.

`osTimerNew` is the CMSIS-RTOS2 API for creating timer objects. The return value should be the timer identifier when everything works fine. The return value 0 is used in case if an error that prevented the timer to be created. The possible causes are highly implementation dependent.

Which RTOS implementation are you using? RTX5? Please consider appending your config files. Add a reproducer example if feasible.

Cheers,
Jonatan

Pupil2013, 2018-03-16:
Thanks @JonatanAntoni ,
I will test my code later with a simpler application .
 Yes ,I'm using CMSIS-RTOS2 API implementated by RTX5

JonatanAntoni, 2018-03-16:
RTX5 has a couple of options on how objects are created, i.e.
- global dynamic storage
- object specific dynamic storage
- user provided (static) storage

Please find the according documentation at http://arm-software.github.io/CMSIS_5/RTOS2/html/theory_of_operation.html#MemoryAllocation. Most likely not enough dynamic storage is available to allocate the needed timer control block from. Inspect your memory pools.

Pupil2013, 2018-03-21:
[Test.zip](https://github.com/ARM-software/CMSIS_5/files/1832063/Test.zip)
hi @JonatanAntoni   the timer still can't work. I upload a simple project.

ReinhardKeil, 2018-04-03:
Hi @Pupil2013,

your program is actually working fine.  You might have been mislead by debugger which does not show timer_id right at the line #28 as the code that stores the value has been re-arranged.

There are two solutions:
  - reduce optimization during debugging to -O1
  - use from the menue View - Watch Windows - RTX RTOS and check the Timer creation using the object viewer.

It is definitely no a problem of RTX5.

JonatanAntoni, 2018-04-09:
Hi @Pupil2013,

as Reinhard already mentioned the creation of the timer in your example seems to work fine.

Unfortunately you try to start the timer before `osKernelStart` which returns an error and does not actually start the timer. Try to move the `osTimerStart` call into your thread `app`. 

Furthermore you must not use blocking calls in your timer callbacks! Please remove `osDelay` from `Timer_Callback` and consider this callback function similar to an interrupt handler function. 

Cheers,
Jonatan

ReinhardKeil, 2018-04-23:
closing it now as there is no further feedback/questions.

closed 2018-04-23 by ReinhardKeil