ARM-software/CMSIS_5 #338: Bug in memory allocation of RTX5?

qwerty19106, 2018-03-19:
I studied the implementation of memory allocators in RTX5 and found it (file rtx_memory.c: 185):
```C
// Memory block header
  ptr = MemBlockPtr(block, 0U);
  ptr--;
```
In this place I expected to see something like this:
```C
// Memory block header
  ptr = MemBlockPtr(block, 0U);
  ptr -= sizeof(mem_block_t);
```
But `sizeof(mem_block_t)` is 2 bytes. Why source code contains `ptr--`, which decrease `ptr` on 1 bytes?

JonatanAntoni, 2018-03-19:
Hi @qwerty19106,

Thanks for getting in touch.

ptr is of type mem_block_t* which in turn is a pointer to struct mem_block_s. Inspecting sizeof(mem_block_t) should give you 8 (assuming 4-byte pointers).

Please make yourself familiar with pointer arithmetic in C. Arithmetic operations on pointer types take the size of the underlying type into account. Hence ptr-- actually decreases the address at 8 bytes not 1 byte. In turn ptr -= sizeof(mem_block_t) would decrease the address at 8*8 = 64 bytes. 

You can easily test this with:
```
	mem_block_t mem_block[9U];
	mem_block_t* ptr1 = &(mem_block[8U]);
	mem_block_t* ptr2 = &(mem_block[8U]);
	ptr1--;
	ptr2 -= sizeof(mem_block_t);
	assert(ptr1 == &(mem_block[7U])); // pointer to one element less
	assert(ptr2 == &(mem_block[0U])); // pointer to eight elements less
```

Cheers,
Jonatan

qwerty19106, 2018-03-19:
Indeed, I did not take pointer arithmetic in C into account.
But the same code as above is written using the function MemBlockPtr (file rtx_memory.c: 155). 
```C
ptr = MemBlockPtr(p, sizeof(mem_block_t));
```
This confused me. Perhaps it would be better to use unified approach to pointer arithmetic.
Thanks for your reply!

closed 2018-03-19 by qwerty19106