ARM-software/CMSIS_5 #339: arm_cmplx_mag_q31 gives strange results

escherstair, 2018-03-19:
I've just found a strange behavior in my application and I'm not so convinced about results from `arm_cmplx_mag_q31()`.
I have the following complex q31_t vector
`source[4] = {-745484, -160984, 375531, 81118}`
and I want to calculate the magnitude.

Based on `arm_cmplx_mag_q31()` documentation

> The function implements 1.31 by 1.31 multiplications and finally output is converted into 2.30 format.

and so I expect to get the following magnitudes
`{381334, 192096}`
Could you confirm these results, please?

Unfortunately, the following results are returned from `arm_cmplx_mag_q31()`:
`{379317, 185363}`
I think the reason is somehow related to an high approximation in `arm_sqrt_q31()`.

The difference between the "expected" results and what I get is too high for my needs.
Could you verify if either `arm_cmplx_mag_q31()` is buggy or can be improved?

escherstair, 2018-03-20:
> I think the reason is somehow related to an high approximation in arm_sqrt_q31().
Or maybe in the right shift
```
/* get the result to 3.29 format */
out1 = (q31_t) (mul1 >> 33);
...
```
which looses too much precision with the supplied values (if I understand well).

llefaucheur, 2018-03-20:
Hi escherstair, the proposed future APIs will include and extra parameter asking for the minimum mumber of valid bits in the mantissa (including trigonotric and hyperbolic functions).
Let's wait a bit (not before 3Q18)

escherstair, 2018-03-20:
Good news that an improved version of DSP library will be implemented.
Bad news that I should wait until Q18 (at least).

escherstair, 2018-10-02:
Hi @llefaucheur 
4Q18 has just started.
Do you have news on the release of the proposed future APIs your write in your message above?

christophe0606, 2020-01-23:
It is not the most accurate implementation but it is not an outlier, for accuracy, compared to other CMSIS-DSP functions.

I have made a few tests:
1 - Adding more iterations to the fix point algorithm used in arm_sqrt_q31 is not improving the accuracy
2 - Keeping more bits after the multiply (>>32 instead of >>33) is not improving things a lot.

So, improving the accuracy of this algorithm will require a bigger rework and since it is not an outlier compared to other CMSIS-DSP function, we are going to keep it as it is.

closed 2020-01-23 by christophe0606

escherstair, 2020-01-23:
@christophe0606 sorry but I can't agree to this.
Two years ago (not two months) @llefaucheur  wrote
> the proposed future APIs will include and extra parameter asking for the minimum mumber of valid bits in the mantissa (including trigonotric and hyperbolic functions).
> Let's wait a bit (not before 3Q18)

Now we are in 1Q20 and instead of the new APIs the issue is closed without any modification?
Frankly speaking, it's not the effort I would expect from ARM...

llefaucheur, 2020-01-23:
Sorry we had to descope this API compared to other urgent tasks like the new Helium architecture MVE, Neon porting, 64bits architectures, CMSIS-NN enhancements, and many more functions you will see in the coming weeks (sort, Gortzel DFT, Splines, Logical operators, float64, new IIR filters, ..).
And up to now we reconsidered the design of new APIs to preserve the legacy ones for trigonometric/hyperbolic functions.