ARM-software/CMSIS_5 #340: Suggestion to add bit- macros for EXC_RETURN/Integrity Signature flags

ioannisg, 2018-03-21:
As a developer, implementing **exception & fault handling** functions for Cortex-M MCUs, I would like to have standard (CMSIS) bit-mask macro #defines for EXC_RETURN flags, namely, ES, SPSEL, Mode, S, Ftype etc., so I would not have to define those in OS/application source code. 

Particularly for ARMv8-M with Security Extension: I would also like to have a macro definition for the value of the Integrity Signature.

JonatanAntoni, 2018-03-21:
Hi @ioannisg,

Thanks for getting in touch about this.
At the first glance your suggestion looks reasonable to me. Are you willing to collaborate? A pull-request with the proposed enhancements is very appreciated.

Cheers,
Jonatan

ioannisg, 2018-03-21:
@JonatanAntoni Yes, I can work on an PR for this one. Will keep you posted.

JonatanAntoni, 2018-03-21:
Great. That would be very helpful as team capacity is always the limiting factor. Thanks.

ReinhardKeil, 2018-05-28:
We added added defines to CMSIS-Core(M) for EXC_RETURN values.

Sorry we have not found the value of Integrity Signature (is this documented somewhere? why is this value important as it is checked by HW only?)

salkinium, 2018-05-28:
> is this documented somewhere?

In the ARMv8-M ARM. Quite literally.

[image]

ioannisg, 2018-05-29:
@ReinhardKeil thanks a lot for the update; the reasoning behind having the needs to check the Integrity Signature in Software could be as follows:

Assume a Cortex-M33 with Security Extensions.
In a SecureFault handler the developer needs to handle SecureFault exception triggered by Non-Secure code.
It is straightforward to dump the Non-Secure stack information. However, the developer may want to also dump the Secure stack information, too.
Now, if the fault has occurred during a Non-Secure exception which has interrupted the Secure execution, the Secure state has stacked the additional state context and the top of the stack contains the **integrity signature**. [In case of a Non-Secure function call the top of the stack contains the return address to Secure state.]
The integrity signature is needed, in order to see if there is also the additional context.

JonatanAntoni, 2018-06-06:
Hi @ioannisg,

can you confirm that the enhancements provided are functional and fulfill your requirements?
If so, please be so kind and close this issue. 

Cheers,
Jonatan

flit, 2018-06-21:
@JonatanAntoni It would be best if `EXC_INTEGRITY_SIGNATUR` could be spelled as `EXC_INTEGRITY_SIGNATURE`. Saving a single character is not worth an abnormal spelling. 😄 Thanks!

JonatanAntoni, 2018-06-22:
Hi @flit, 

Thanks for pointing this out.
We will spent this _huge_ amount of additional disk space ;)

Cheers,
Jonatan

JonatanAntoni, 2019-03-12:
Hi @ioannisg, May I ask you to cross check the provided changes and finally close this issue, please?

ioannisg, 2019-03-12:
> 
> 
> Hi @ioannisg, May I ask you to cross check the provided changes and finally close this issue, please?

Looks good, sorry for delaying this

closed 2019-03-12 by ioannisg