ARM-software/CMSIS_5 #342: Question about RTX5 Memory Pool implementation

qwerty19106, 2018-03-23:
I studied the implementation of memory pools in RTX5. And I have questions about the following code (file rtx_mempool.c: 93):

```
  __disable_irq();

  if (mp_info->used_blocks < mp_info->max_blocks) {
    mp_info->used_blocks++;
    block = mp_info->block_free;
    if (block != NULL) {
      //lint --e{9079} --e{9087} "conversion from pointer to void to pointer to other type"
      mp_info->block_free = *((void **)block);
    }
  } else {
    block = NULL;
  }

  if (primask == 0U) {
    __enable_irq();
  }
```

In which case the condition `mp_info->used_blocks < mp_info->max_blocks` is satisfied,
while `mp_info->block_free == NULL`?

qwerty19106, 2018-03-30:
It seems I found the reason for such a strange check.
Since no one answered me, I'll try to answer myself, to help the community and fix one bug.
I hope you know the word **sarcasm**!

I think this condition came from atomic case in osRtxMemoryPoolAlloc (rtx_mempool.c: 111):
```C
  if (atomic_inc32_lt(&mp_info->used_blocks, mp_info->max_blocks) < mp_info->max_blocks) {
    block = atomic_link_get(&mp_info->block_free);
  } else {
    block = NULL;
  }
```
and atomic case in osRtxMemoryPoolFree (rtx_mempool.c: 157):
```C
  if (atomic_dec32_nz(&mp_info->used_blocks) != 0U) {
    atomic_link_put(&mp_info->block_free, block);
    status = osOK;
  } else {
    status = osErrorResource;
  }
```
In fact code with disabled IRQ `#if(EXCLUSIVE_ACCESS == 0)` repeat atomic code.
Consider this case: 
1) pool have 0 free blocks, `mp_info->used_blocks == mp_info->max_blocks`
2) In IRQ mode occur `osRtxMemoryPoolFree` call
3) `atomic_dec32_nz` is completed, `mp_info->used_blocks == mp_info->max_blocks - 1`, but at the same time `mp_info->block_free == NULL` yet.
4) occur IRQ with higher priority, which call `osRtxMemoryPoolAlloc`!
5) Call `atomic_inc32_lt` (rtx_mempool.c: 111) increment mp_info->used_blocks
6) Call `atomic_link_get(&mp_info->block_free)`, but `mp_info->block_free == NULL` yet!

I guess that to prevent it you add `mp_info->block_free == NULL` check to asm code in atomic_link_get (rtx_core_cm.h: 1118). Then anologious check was added to non atomic case. But it is fight with windmills, and not with the mistake itself.
**Result**: `osRtxMemoryPoolAlloc` incremented `mp_info->used_blocks`, but it did not take block from the pool!
**Facepalm..**

I offer this code for `osRtxMemoryPoolFree` (although this code is also not optimal, but it is not subject to the described error): 
```C
  atomic_link_put(&mp_info->block_free, block);
  if (atomic_dec32_nz(&mp_info->used_blocks) != 0U) {
    status = osOK;
  } else {
    status = osErrorResource;
  }
```

If we consider the problem wider:
Why we  check `mp_info->used_blocks`, while `mp_info->block_free` reflects the real state of the pool?
If `mp_info->block_free != NULL`, we can get block from pool. We can to put block in pool any time.
As long as we not check double call of `osRtxMemoryPoolFree` for one block, the `mp_info->used_blocks` check does not make sense.

Wherein `mp_info->used_blocks` not consisted really count of free blocks. It might be useful to add function `uint32_t osMemoryPoolIsEmpty(osMemoryPoolId_t mp_id)`, which return `mp_info->block_free == NULL`.

Sincerely, Krivenkov R.

RobertRostohar, 2018-04-03:
You are right. There is no need for the double check of `used_blocks` and `block_free` in the `osRtxMemoryPoolAlloc` function. Actually in the atomic case the double check has a race condition as you pointed out.

The code has been updated and now only checks the `block_free` when allocating.

Thanks for the analyses and feedback.

JonatanAntoni, 2018-04-03:
Hi @qwerty19106,

Many thanks for your valuable feedback. Please excuse the delay due to vacation period.

May I ask you to double check the fix Robi provided, please? Please close the issue if your concerns are solved.

Cheers,
Jonatan

qwerty19106, 2018-04-06:
I think this code is good.
Now a problems can arise only in the case of double free of one block. But for such implementation of the pool it seems to impossible to check this case.
Issue is closed.

Sincerely, Krivenkov R.

closed 2018-04-06 by qwerty19106

JonatanAntoni, 2018-04-09:
Thanks for your conformation.
Double free is an issue we have under discussion as well. It might be possible to detect double free's with additional overhead. Hence we decided to put this difficulty to the documentation for the time being.