ARM-software/CMSIS_5 #343: Importing weights from a model trained in Tensorflow to CMSIS NN

petitcoeur9000, 2018-03-29:
Hi,

I am interested in training a CNN in Tensorflow and then importing the trained weights to a CNN implemented with the CMSIS NN library. Is it doable? Is there any need to preprocess the weights from the Tensorflow model before importing it?

Thanks

liangzhen-lai, 2018-04-03:
Hi @petitcoeur9000 ,

Yes, it is doable. We have a few examples of model trained in tensorflow and ported to the CMSIS-NN library. It will require quantization on the weights and activations. You can refer to this for more details:
https://github.com/ARM-software/ML-KWS-for-MCU/blob/master/Deployment/Quant_guide.md

We are also working on some simpler (e.g., semi-automatic) ways for people to port Tensorflow model with CMSIS-NN. For now, you can refer to the documents mentioned above. Let me know if this answers your question.

Thanks,
Liangzhen

petitcoeur9000, 2018-04-04:
Thanks for the answer! the tensorflow porting examples you are referring to are all contained in the ML-KWS-for-MCU repo or are they also somewhere else?

liangzhen-lai, 2018-04-05:
Also, if you have fully-connected layers in your network and want to use the kernel with weight reordering (i.e., arm_fully_connected_**_opt), you can refer to this script for weight reordering:
[Script link](https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/NN/Scripts/NNFunctions/fully_connected_opt_weight_generation.py)

lishunde, 2018-04-09:
It's great to see this issues,I also want to know how to generate the data in  [https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/NN/Examples/ARM/arm_nn_examples/cifar10/arm_nnexamples_cifar10_shift.h](url)

petitcoeur9000, 2018-04-10:
Hi @lishunde ,

see #327 to understand how to generate the shift header.

fengzi0, 2018-04-18:
Hi @liangzhen-lai ， i'm trying to CNN model from TensorFlow to CMSIS_NN，but the order of CNN's weight is puzzling me，is it N(output_channel)H(eight)W(idth)C(input_channel)? or HWCN as TensorFlow? Also about full connection layer, is it N(output_channel)C(input_channel) or CN?

liangzhen-lai, 2018-04-18:
Hi @fengzi0 , it should be NHWC and NC.

sbochkar, 2018-04-25:
Hi @liangzhen-lai ,

Just want to clarify the CNN's weight format that arm_convolve_HWC_q7 expects.

Does it expect a flattened array of weights where the first index varies the slowest and the last index varies the fastest?

Example:
Suppose I have 29 5x5 conv. filters acting on 3 input channels.
In Caffe this would be an array of shape (29, 3, 5, 5). NCHW
To make this work with CMSIS-NN, I should first rearrange this matrix into (29, 5, 5, 3) [NHWC]. And then flatten using row major transform. e.g. numpy.ravel(). (e.g. last index varies fastest)

Can you confirm this is correct? Thank you very much for your time.

liangzhen-lai, 2018-04-25:
@sbochkar , Yes, the weight ordering is correct. In the cnn example in CMSIS-NN, we use `numpy.reshape([N*H*W*C])` to flatten the weights. I guess `numpy.ravel()` should behave the same.

iabdalkader, 2018-05-03:
@liangzhen-lai  Hi, I'm also trying to import Caffe models. Can you share the Caffe model you used in the example so I have something to compare to ?

iabdalkader, 2018-05-11:
>> We are also working on some simpler (e.g., semi-automatic) ways for people to port Tensorflow model with CMSIS-NN.

@liangzhen-lai  Actually never mind the model, have you made any progress on this ?

yyz1985, 2018-10-24:
> Also, if you have fully-connected layers in your network and want to use the kernel with weight reordering (i.e., arm_fully_connected_**_opt), you can refer to this script for weight reordering: [...]

what if n is not multiply of 4, like 10?

csukuangfj, 2019-01-30:
@yyz1985

> what if n is not multiply of 4, like 10?

It works as you expected. Please read the code for more details.