ARM-software/CMSIS_5 #345: support removal of register storage class specifier in C++17

pabigot, 2018-04-02:
Prior to C++17 the `register` storage class specifier was deprecated; in C++17 it has been removed, making attempts to compile with CMSIS 5.3 with gcc `--std=c++17` fail.  Something like the following is needed in every header that uses that feature, along with the corresponding modifications at all use-points.

    #if (__cplusplus - 0) >= 201703L
      #define __REGISTER
    #else
      #define __REGISTER                             register
    #endif

See: http://en.cppreference.com/w/cpp/language/storage_duration

JonatanAntoni, 2018-04-09:
May I ask you to be slightly more specific, please?
Where exactly do you have this issue?

Please be aware that CMSIS itself it intended to be ANSI-C (C99) and not C++. Simply removing the storage class `register` from the code is not guaranteed to work. Depending on the implementation it might be required that a value is kept as a register value only.

Using CMSIS from C++ should be possible when compiling all .c files in C-Mode. Inline functions defined in header files should be compiled as `extern C`. Please let me know if that does not work for you.

pabigot, 2018-04-09:
As of C++17 the existing solution does not work.

CMSIS Core headers like `core_cm0.h` can be included into C++ source files; this is enabled through the use of the `__cplusplus` macro to add the required `extern "C"` block around the declarations.  This was sufficient to allow core inline functions like `NVIC_SetEnableIRQ()` and structures like `SCB_Type` to be available to and used in C++ applications through C++14.

The change to the `register` storage class specifier in C++17 means this isn't possible anymore because simply including `core_cm0.h` into a C++ application pulls in `cmsis_compiler.h` which pulls in `cmsis_gcc.h` which declares a local `register uint32_t result;` which is a syntax error.  The modification I suggest is required if the existing capability is to be maintained beyond C++14.

The change can probably be conditional on using C++ at an affected version (as suggested) and isolated to compiler-specific headers like `cmsis_gcc.h` which should instead declare `__REGISTER uint32 result;`. For implementation files that use register declarations in C99-compiled code I agree it's not relevant and undesirable.

Some information on the reason for the change is [here](https://stackoverflow.com/a/30809775/236848). As that material indicates gcc completely ignores the `register` keyword unless all optimizations are explicitly disabled (the `__ASM` statements in CMSIS include a constraint that ensures a register is used).  In practice I've had no problems with gcc-arm-none-eabi-7-2017-q4-major and `--std=c++17` with this modification.

JonatanAntoni, 2018-04-09:
Thanks for the details. I can know comprehend the issues you face. Its all about the inline functions using raw assembly instructions for special function register accesses like

```
__STATIC_FORCEINLINE uint32_t __get_PSP(void)
{
    register uint32_t result;
    __ASM volatile ("MRS %0, psp"  : "=r" (result) );
    return(result);
}
```

In other function we already skipped the `register` keyword completely. To be consistent we could remove the `register` keyword from `cmsis_gcc.h` completely. The same change should apply to `cmsis_armclang.h`. 

Is it feasible for you to provide the changes in terms of a pull request?

Thanks again for your contribution.

pabigot, 2018-04-09:
Yes, I can do a PR sometime in the next week or so.  Not sure whether I'll be able to test cmsis_armclang but if not will note that in the PR.

FreddieChopin, 2018-05-11:
Alternative pull request which removes the keyword wherever possible -#365

JonatanAntoni, 2018-05-14:
Thanks @FreddieChopin for your contribution.
May I ask you all to double check the result, please?

pabigot, 2018-05-22:
@JonatanAntoni @FreddieChopin The merged changes work for my application.  Thanks.

closed 2018-05-22 by pabigot