ARM-software/CMSIS_5 #349: Constness of input parameters for CMSIS-DSP functions

escherstair, 2018-04-12:
Some functions of CMSIS-DSP library take vectors as input parameters (passed by pointer to the first element).
Some of these functions don't modify at all these vectors and so it would be better if the type of the input parameters is `const _type_* pSrc`.
As an example `arm_max_xxx()`, `arm_mean_xxx()`, `arm_min_xxx()`, and so on.
For other functions it's not clear if they can work in place by design (and so input vector cannot be pointer-to-const) or not (and so input vector could be declared as pointer-to-const).
As an example,I refer to `arm_q31_to_float()` and so on.

Could you review the declarations of the functions in CMSIS-DSP library and set the constness when needed?

llefaucheur, 2018-04-13:
Hi, the current direction is 1) keep API as-is for legacy of customer code investments (including Mathworks)    2) add blas API    3) prepare the path to Cortex-A/R devices with vector API with strides (for example arm_vabs_TYP (TYP *pA, s32 stridesA, real_TYP *pOut, s32 stridesOut, s32 N, ctrl *S), TYP in {q7,q15,q31, .. zq7,zq15,zq31(complex) , f8,f16,f32.. zf8,zf16,zf31..}). The *ctrl being there for future extensions (circular buffer addressing, ..).

escherstair, 2018-04-16:
Hi @llefaucheur 
sorry but I don't understand your answer.
Could you clarify it, please?

escherstair, 2018-06-22:
In `arm_max_q31(q31_t *pSrc, uint32_t blockSize, q31_t *pResult, uint32_t *pIndex)`, **`pSrc`** should be declared as `const q31_t*` too.

anton-dirac, 2019-03-05:
Why was this reverted?

JonatanAntoni, 2019-03-05:
Hi @anton-dirac,

the provided patch broke the library.

Cheers,
Jonatan

anton-dirac, 2019-03-05:
Ok! That's a good reason to revert it :)

escherstair, 2021-07-07:
Do you have news on this task?

christophe0606, 2021-07-07:
@escherstair It has been progressively added to the library.
If you see missing cases, thanks to tell me. I'll improve it.

christophe0606, 2022-06-30:
We have been continuously improving the library for `const` correctness.
I consider this task mainly done. I am not saying it is perfect and there may be few remaining cases where `const` is missing. In that case, new github issues will be created.

PR #531 (2019-02-14): Const-qualify all read-only pointers in CMSIS-DSP (#349)
This PR closes #349. The lack of const-correctness has been bugging me lately as I've been writing a C++ wrapper around the CMSIS-DSP library, and my C++ linter and compiler are quite strict in requiring const correctness. 

I've also fixed a couple of recurring typos. Adding `const` might in some cases help the compiler optimize further as well, but this has not been verified. 

In some source files, a pointer has been reused to point to both a `const` and non-`const` field. In these cases, I've elected to do a cast from `const` to non- [... cut]
closed 2022-06-30 by christophe0606