ARM-software/CMSIS_5 #356: [CMSIS-NN] Pooling functions for q15_t?

sbochkar, 2018-04-24:
Hello everyone,

I'm trying to implement my CNN on a Cortex-M7 using CMSIS-NN kernels. I trained my model in Caffe on the host computer. As such, my weights are all floating points. Hence, I try to quantize these values into q7 format as per examples.

However, my network does not work. I suspect that the quantization into 8 bits does not provide me with enough resolution. I would like to try quantizing into 16 bits. I see most of the kernels support q15_t datatypes. However, there is no pooling function for q15_t type.

Is there one coming for q15_t support or is there a workaround for this?

Thanks a lot.

liangzhen-lai, 2018-04-25:
Hi @sbochkar,

Thank you for raising the issue. As a _temporary_ workaround, you can implement your own q15_t pooling function. We do have a reference implementation in arm_pool_q7_HWC.c, maxpool from line 254 to 285 and avgpool from line 412 to line 442. It should be relatively straightforward to change them to implement q15_t pooling. I have an example at the bottom.

 Let me know if this solves your problem. 

Thanks,
Liangzhen

Below is an example for q15_t maxpool:

```
void arm_maxpool_q15_HWC_ref(q15_t * Im_in,
                   const uint16_t dim_im_in,
                   const uint16_t ch_im_in,
                   const uint16_t dim_kernel,
                   const uint16_t padding,
                   const uint16_t stride, const uint16_t dim_im_out, q15_t * bufferA, q15_t * Im_out)
{
    int16_t   i_ch_in, i_x, i_y;
    int16_t   k_x, k_y;

    for (i_ch_in = 0; i_ch_in < ch_im_in; i_ch_in++)
    {
        for (i_y = 0; i_y < dim_im_out; i_y++)
        {
            for (i_x = 0; i_x < dim_im_out; i_x++)
            {
                int       max = -32769; // i.e., -1 * (2^15+1) so that it is smaller than smallest q15_t
                for (k_y = i_y * stride - padding; k_y < i_y * stride - padding + dim_kernel; k_y++)
                {
                    for (k_x = i_x * stride - padding; k_x < i_x * stride - padding + dim_kernel; k_x++)
                    {
                        if (k_y >= 0 && k_x >= 0 && k_y < dim_im_in && k_x < dim_im_in)
                        {
                            if (Im_in[i_ch_in + ch_im_in * (k_x + k_y * dim_im_in)] > max)
                            {
                                max = Im_in[i_ch_in + ch_im_in * (k_x + k_y * dim_im_in)];
                            }
                        }
                    }
                }
                Im_out[i_ch_in + ch_im_in * (i_x + i_y * dim_im_out)] = max;
            }
        }
    }
}
```

JonatanAntoni, 2018-05-08:
Hi @sbochkar,

please consider closing this issue if the provided solution successfully answers your question.

Thanks,
Jonatan

closed 2018-05-08 by sbochkar