ARM-software/CMSIS_5 #361: What's the meaning of 'multiple-of-4 rows' refferde in arm_fully_connected_q7_opt.c

lishunde, 2018-05-09:
[image]
It's hard for me to understanded the weight convert style in [arm_fully_connected_q7_opt.c](url)
And I'm import the weight from tensorflow to CMSIS-NN ,which have the shape (10,3136) . 
with the sam input data,I've set all of the biases to 0,but the output always be different .I thought it caused by the False corder of the weight.isn't it?

lishunde, 2018-05-10:
@liangzhen-lai   I've make a test as shown in the following description

> input:  same  , tensorflow is float16,   cmsis-nn is Quantize to q7_t

>weight:  transform by using the scripe ,and Quantized to q7_t

> biases:  zero
and I  test it by using `tf.matmul(input, weights)`and`tf.nn.relu(tf.matmul(reshape, weights)'respectively ,
but the output  error is relatively  bigger,  May My calculation is not correct.?

liangzhen-lai, 2018-05-10:
Hi, @lishunde ,

Can you check if the dimension ordering is correct? You can refer to this issue for more details #343 .

Thanks,
Liangzhen

liangzhen-lai, 2018-05-11:
Can you try the arm_fully_connected_q7 version first? That should be working without weight reordering.
You can also take a look at this [https://github.com/ARM-software/CMSIS_5/tree/develop/CMSIS/NN/NN_Lib_Tests/nn_test/Ref_Implementations](url) , where we have the reference implementation.

lishunde, 2018-05-11:
I also make a test by using arm_fully_connected_q7 and arm_fully_connected_q7_ref as it's usage in nn_test.
But the  precision  is also quite different.
The fullly-connect layer make C times multiplication per output  ,it's more larger than conv. isn't the round-off accumulating  cause the error.

liangzhen-lai, 2018-05-15:
Hi @lishunde ,

Sorry I don't quite understand your question here. Do you mean that the results generated by `arm_fully_connected_q7` and `arm_fully_connected_q7_ref` are different for the same inputs?

Thanks,
Liangzhen

lishunde, 2018-05-20:
@liangzhen-lai No,I compared the output data bwtween the `arm_fully_connected_q7 or arm_fully_connected_q7_ref`and the tensorflow code.

felix-johnny, 2021-01-07:
@lishunde Some late tidying up of tickets. The output of _q7 API's cannot be compared with tensorflow as there are multiple differences between them. Please have a look at https://github.com/ARM-software/CMSIS_5/tree/develop/CMSIS/NN#legacy-vs-tfl-micro-compliant-apis for more details. 
 Hope that explains.

closed 2021-01-07 by felix-johnny