ARM-software/CMSIS_5 #362: What's the meaning of 'NN_ROUND(out_shift)' refferde in arm_convolve_HWC_q7_RGB.c

baileo, 2018-05-10:
Why the bias add NN_ROUND(out_shift)?

liangzhen-lai, 2018-05-10:
Hi @baileo ,

This is used to support different rounding mechanisms, e.g., floor vs. rint.

Let me know if this answers your question. Please consider closing the issue if you do not have any follow-up questions.

Thanks,
Liangzhen

Dustin-Huang, 2018-05-11:
Hi Liangzhen,
Would please say more detail about  NN_ROUND(out_shift)?
why supporting different rounding mechanisms must add the 2**(out_shift-1)?
Thanks

ablaylock, 2018-05-11:
You too Mr pig

liangzhen-lai, 2018-05-15:
There are different ways to quantize a fractional number into an integer. For example, floor vs. ceil vs. rint. 
Since floor(n) = rint(n+0.5), to facilitate different rounding options, we use this "NN_ROUND" macro is used to indicate the value adjustments, i.e., 0.5 or 2**(out_shift-1).

JonatanAntoni, 2018-06-27:
@Dustin-Huang, does this answer solve your question? Please consider closing this issue.

JonatanAntoni, 2019-03-12:
Closing due to inactivity.

closed 2019-03-12 by JonatanAntoni