ARM-software/CMSIS_5 #364: Support large FFT lengths for arm_rfft_fast_f32

WBABD, 2018-05-11:
I think support for FFTs up to 16,384 would be very useful for general use.  
 I understand there would be a huge code size hit due to the twiddleCoef array needed but perhaps there could be a makefile define to enable this functionality. 

Thank you for this fantastic library anyway.

christophe0606, 2019-05-21:
We are currently working on some changes which will allow to only include the needed tables in CMSIS-DSP. So, it should be possible to support bigger FFT lengths without impact on the code size since if the table is not needed it won't be included.

WBABD3, 2019-05-21:
Thank you christophe0606

alexheslop, 2019-11-27:
Hi @christophe0606, what is the current maximum FFT size available?

christophe0606, 2020-01-23:
Improvements to the FFT will require a full rework since we need to address many different requests and needs.

So, this won't be fixed in current implementation but we keep track of the need and request for future.

closed 2020-01-23 by christophe0606