ARM-software/CMSIS_5 #366: feature request: nonsquare q15 convolve

quinnabrvau, 2018-05-13:
Hi,
I have an application that would benefit from an arm_convolve_HWC_q15_(fast or basic)_nonsquare
Thank You

liangzhen-lai, 2018-05-24:
Hi @quinnabrvau ,

The requested function is included in the pull request #372 . Let me know if you find any problem using that.

Thanks,
Liangzhen

JonatanAntoni, 2018-06-27:
Closing due to inactivity. Please feel free to reopen if the issue still exists.

closed 2018-06-27 by JonatanAntoni

iabdalkader, 2018-10-19:
@JonatanAntoni @liangzhen-lai Shouldn't the pooling functions also handle non-square images ?

JonatanAntoni, 2018-10-19:
Hi @iabdalkader,

Sounds reasonable to me on the first glance. 
Please raise this as a separate enhancements request, i.e. which additional functions would you like to see? Providing implementation proposals for these functions in terms of a pull-request is highly appreciated.

Thanks,
Jonatan