ARM-software/CMSIS_5 #368: A script to import Caffe models.

iabdalkader, 2018-05-14:
Hi, I'm writing a script to convert Caffe models to CMSIS NN format. It's still a work in progress (some steps are manual) but I'm getting there. I've been gathering bits and pieces of information from here and there but it seems I'm still missing something and would appreciate any help I can get. Here's what I have so far, I trained and tested the _quick_ model (~71% accuracy) after removing the IP2 layer to match the example as much as possible.

## Quantize weights:
I wrote a script to load the network, quantize the weights (using the function in [quantization guide](https://github.com/ARM-software/ML-KWS-for-MCU/blob/master/Deployment/Quant_guide.md#quantize-weights
)) and write back C arrays. The important bits:

### Reshaping weights:
```
if (ltype == 'Convolution'):
    # reshape to N, H, W, C
    weights = weights.transpose((0, 2, 3, 1))
```
Note I use nditer later:
```for i in np.nditer(l['w']):```

### The quantization function:
```
def quantize_weights(layer, weights):
    min_wt = weights.min() 
    max_wt = weights.max()
    # Find number of integer bits to represent this range
    int_bits = int(np.log2(np.ceil(max(abs(min_wt),abs(max_wt))))) 
    frac_bits = 7-int_bits # Remaining bits are fractional bits (1-bit for sign)
    print("%s int bits %d frac_bits %d"%(layer, int_bits, frac_bits))
    return np.round(weights*(2**frac_bits)).astype(np.int8)
```
_Note I don't reorder IP weights for now just using the `arm_fully_connected_q7` function._

### The output format for weights and biases:
```
conv1 :int bits 0 frac_bits 7
conv1 :int bits 0 frac_bits 7
conv2 :int bits 0 frac_bits 7
conv2 :int bits 0 frac_bits 7
conv3 :int bits 0 frac_bits 7
conv3 :int bits 0 frac_bits 7
ip1   :int bits 0 frac_bits 7
ip1   :int bits 0 frac_bits 7
```

## Quantize activations:
I used this [script](https://github.com/NVIDIA/DIGITS/blob/master/examples/classification/example.py) to test the network on sample images, and find the output format.
### Calculate shift bits
The output format for each layer
```
conv1 :int bits 7 frac_bits 0
conv2 :int bits 7 frac_bits 0
conv3 :int bits 6 frac_bits 1
ip1   :int bits 3 frac_bits 4
```
The shift bits are calculated like this (assuming the input is Q7.0) :
```
def calc_shifts(i, w, b, o):
    iw = [31 - (i[1] + w[1]), (i[1] + w[1])]
    bshift = iw[1] - b[1]
    oshift = iw[1] - o[1]
    return (bshift, oshift)

print(calc_shifts((7, 0), (0, 7), (0, 7), (7, 0)))
print(calc_shifts((7, 0), (0, 7), (0, 7), (7, 0)))
print(calc_shifts((7, 0), (0, 7), (0, 7), (6, 1)))
print(calc_shifts((6, 1), (0, 7), (0, 7), (3, 4)))
```

### The bias_shift and output_shift respectively:
```
(0, 7)
(0, 7)
(0, 6)
(1, 4)
````

## Updating the example
I just update the weights arrays and the layer info and shift bits, without changing anything else (notice the output channels from CONV3 is 64 Vs 32 for the CMSIS example, and POOL2/3 padding is 1):

```
#define CONV1_IM_CH         (3)
#define CONV1_IM_DIM        (32)
#define CONV1_KER_DIM       (5)
#define CONV1_PADDING       (2)
#define CONV1_STRIDE        (1)
#define CONV1_OUT_CH        (32)
#define CONV1_OUT_DIM       (32)

#define POOL1_KER_DIM       (3)
#define POOL1_PADDING       (0)
#define POOL1_STRIDE        (2)
#define POOL1_OUT_DIM       (16)

#define CONV2_IM_CH         (32)
#define CONV2_IM_DIM        (16)
#define CONV2_KER_DIM       (5)
#define CONV2_PADDING       (2)
#define CONV2_STRIDE        (1)
#define CONV2_OUT_CH        (32)
#define CONV2_OUT_DIM       (16)

#define POOL2_KER_DIM       (3)
#define POOL2_PADDING       (1)
#define POOL2_STRIDE        (2)
#define POOL2_OUT_DIM       (8)

#define CONV3_IM_CH         (32)
#define CONV3_IM_DIM        (8)
#define CONV3_KER_DIM       (5)
#define CONV3_PADDING       (2)
#define CONV3_STRIDE        (1)
#define CONV3_OUT_CH        (64)
#define CONV3_OUT_DIM       (8)

#define POOL3_KER_DIM       (3)
#define POOL3_PADDING       (1)
#define POOL3_STRIDE        (2)
#define POOL3_OUT_DIM       (4)

#define IP1_DIM             (4*4*64)
#define IP1_OUT             (10)

#define CONV1_BIAS_LSHIFT   (0)
#define CONV1_OUT_RSHIFT    (7)

#define CONV2_BIAS_LSHIFT   (0)
#define CONV2_OUT_RSHIFT    (7)

#define CONV3_BIAS_LSHIFT   (0)
#define CONV3_OUT_RSHIFT    (6)

#define IP1_BIAS_LSHIFT     (1)
#define IP1_OUT_RSHIFT      (4)
```

This still doesn't work (i.e. accuracy < 25%). Any idea what I'm missing ?

akioolin, 2018-05-15:
we also have the same problem, especially in output / bias shift phase. in the quantization guide, there is no talking about how decide the shifting bits.

liangzhen-lai, 2018-05-16:
Hi @iabdalkader  @akioolin 

We will soon upload the scripts that we used to generate the parameters from a trained caffe model.
Will give the link once it's ready.

Thanks,
Liangzhen

liangzhen-lai, 2018-05-17:
We have uploaded the script and caffe model we used to generate the original cifar10 CNN example (commit f8c5612872b2c021de2a011b63b4f51f613f985e) here:
[https://github.com/ARM-software/ML-examples/tree/master/cmsisnn-cifar10](url)

iabdalkader, 2018-05-19:
@liangzhen-lai Thanks!

I think there's a bug in code_gen.py in line
https://github.com/ARM-software/ML-examples/blob/master/cmsisnn-cifar10/code_gen.py#L214

And line
https://github.com/ARM-software/ML-examples/blob/master/cmsisnn-cifar10/code_gen.py#L215

It should be prev_layer.

```
caffe_model.layer_type[prev_layer]=='convolution' or caffe_model.layer_type[prev_layer]=='4' or\
caffe_model.layer_type[prev_layer]=='pooling' or caffe_model.layer_type[prev_layer]=='17':
```
It doesn't change the buffer size though.

iabdalkader, 2018-05-23:
Hi, found another bug in code_gen.py testing with lenet:
```
Traceback (most recent call last):
  File "code_gen.py", line 362, in <module>
    generate_parameters(my_model, cmd_args.out_dir+'/parameter.h')
  File "code_gen.py", line 94, in generate_parameters
    f.write("#define "+layer.upper()+"_OUT_DIM "+str(caffe_model.layer_shape[layer][2])+"\n\n")
IndexError: tuple index out of range
```
Seems relu doesn't have the expected shape:

sangnguyenz, 2019-05-28:
> Hi, found another bug in code_gen.py testing with lenet:
> 
> ```
> Traceback (most recent call last):
>   File "code_gen.py", line 362, in <module>
>     generate_parameters(my_model, cmd_args.out_dir+'/parameter.h')
>   File "code_gen.py", line 94, in generate_parameters
>     f.write("#define "+layer.upper()+"_OUT_DIM "+str(caffe_model.layer_shape[layer][2])+"\n\n")
> IndexError: tuple index out of range
> ```
> 
> Seems relu doesn't have the expected shape:
> `('relu1', (100, 500))`

I ran into this issue too, I'm deploying mnist on NUCLEO-F746ZG like the tutorial cifar10. Did anyone find the solution? Thanks

rafakath2020, 2019-11-08:
I am stuck with same error.

Traceback (most recent call last):
  File "code_gen.py", line 362, in <module>
    generate_parameters(my_model, cmd_args.out_dir+'/parameter.h')
  File "code_gen.py", line 94, in generate_parameters
    f.write("#define "+layer.upper()+"_OUT_DIM "+str(caffe_model.layer_shape[layer][2])+"\n\n")
IndexError: tuple index out of range

if any one have solution then, kindly provide.