ARM-software/CMSIS_5 #373: Why the weights of CMSIS-NN's CIFAR10 example are not the same as the weights generated by the code generator ?

baileo, 2018-05-25:
https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/NN/Examples/ARM/arm_nn_examples/cifar10/arm_nnexamples_cifar10_weights.h

https://github.com/ARM-software/ML-examples/blob/master/cmsisnn-cifar10/code/m7/weights.h

https://github.com/ARM-software/ML-examples/blob/master/cmsisnn-cifar10/code/m4/weights.h

liangzhen-lai, 2018-05-25:
Hi @baileo ,

As posted also in #368 , the script and caffe model was used to generate the original cifar10 CNN example (commit f8c5612). The latest example is generated using a different flow.

Thanks,
Liangzhen

baileo, 2018-05-30:
What kind of  the different flow?

liangzhen-lai, 2018-05-30:
It's a flow under development that allows one to train the network in Tensorflow with a "push-button" flow for deployment.

majianjia, 2018-12-12:
> 
> 
> It's a flow under development that allows one to train the network in Tensorflow with a "push-button" flow for deployment.

Is there any release after?