ARM-software/CMSIS_5 #377: Information needed

mkamrao, 2018-06-15:
Hi

Could you please suggest about documentation/example usage of any Deep Learning network (such as Alexnet) creation using the CMSIS NN APIs.

Thanks in advance
Kamrao

liangzhen-lai, 2018-06-18:
Hi @mkamrao ,

If I understand your question correctly, you were asking for some materials for creating networks with CMSIS-NN APIs. You can refer to the following link with an example of porting a trained Caffe network to CMSIS-NN implementation:
[https://github.com/ARM-software/ML-examples/tree/master/cmsisnn-cifar10](url)

Let me know if this answers your question. Please consider closing the issue if you don't have any follow-up questions.

Thanks,
Liangzhen

mkamrao, 2018-06-19:
Hi Liangzhen

Thank you, yes this answers my query .

Kamrao

closed 2018-06-19 by mkamrao