ARM-software/CMSIS_5 #378: CMSIS-RTOS Validation Suite for RTOS v2 API

wally4u, 2018-06-20:
According to the documentation the CMSIS RTOS Validation suit is only available publicly for RTOS v1 API.
There appears to be a "private" version for RTOS v2 API. See https://www.keil.com/pack/doc/CMSIS/RTOS2/html/rtosValidation.html  -> "Currently, a public version of the test suite is available only for CMSIS-RTOS v1 API."
It would be appreciated if this is also available publicly since validation is important to me (and probably others).

JonatanAntoni, 2018-06-20:
Hi @wally4u,
thanks for getting in touch with us regarding validation.

What are you exactly looking for? The "CMSIS RTOS Validation" is intended only to proof that an arbitrary implementation actually complies to the API specification to a certain extend. That is far behind an details validation and qualification of a concrete implementation.

Regarding RTX5 (our native implementation of CMSIS-RTOS2) we are currently spending much effort on a [standard compliant safety certification](http://www2.keil.com/safety/rtos). Of course this will not be applicable for other implementations of CMSIS-RTOS2 API.

Best,
Jonatan

wally4u, 2018-06-20:
Hi @JonatanAntoni ,
I understand that the v1 implementation is a limited scope validation test.
I currently use it a sanity test / unit test for the implementations to check changes in the CMSIS packs.
Having a v2 implementation would be beneficial for non MDK-pro users (assuming it exist and requires minimal effort from ARM to put this github).

As an alternative, I'll take a look at the TUV certification since we already have MDK-Professional licenses.

JonatanAntoni, 2018-06-20:
Hi @wally4u,

Do you refer to the API compliance? Or functional test coverage of the actual implementation?
Currently we do not have anything to share on GitHub. Contributions are always appreciated.

Cheers,
Jonatan

wally4u, 2018-06-20:
Hi @JonatanAntoni,
API compliance is a given since without it it defeats the purpose of having an incompatible CMSIS validation test. 
Functional tests coverage such as the CMSIS-RTOS Validation Example Project in the Keil is indeed useful.

Anyway, if there is nothing available to share you can close this issue as a feature request (for github or inclusion in keil uvision pack installer)
Thanks!

JonatanAntoni, 2018-06-20:
Typically I leave it up to the original requester to close issues if satisfied.

The current plan is to provide certification packages for CMSIS, C Library, RTX5, etc. which contains additional stuff (e.g. test suites, documentation) one would need to build safety related systems on top of it. This will not be available free of charge due to the effort it takes.

closed 2018-06-20 by wally4u