ARM-software/CMSIS_5 #380: inverse FFT from 8192 real samples

escherstair, 2018-06-20:
Documentation for `arm_rfft_q31()` states

> Supported FFT Lengths are 32, 64, 128, 256, 512, 1024, 2048, 4096, 8192.

and so I can do a real FFT from a buffer of 8192 samples.
The output is complex and so when I need an inverse-FFT the only option is `arm_cfft_q31()`.
But the documentation states

> The algorithm supports lengths of [16, 32, 64, ..., 4096]

and so how can I do the inverse-FFT when I have a 8192 buffer?

llefaucheur, 2018-06-21:
Sorry, I ckecked this option is not feasible in the library.

closed 2018-07-10 by llefaucheur

llefaucheur, 2018-07-10:
Hi escherstair, yes we will consider this improvement. We seing several directions : improving the API with the removal of implementation details (Radix-4 data..) going beyond 64k points, and reduce the memory consumption for small RAM devices.