ARM-software/CMSIS_5 #381: need suggestion on h/w reference boards

uu-kameshm, 2018-06-22:
Hi

Could you please let me know, on which h/w reference boards/platforms, the CMSIS NN has been already tested.

Basically, I would like to use CMSIS NN on similar reference boards to test applications created using CMSIS NN to eliminate any incompatibility issues.

Thanks
Kamesh

liangzhen-lai, 2018-06-23:
Hi @uu-kameshm ,

CMSIS is designed to be vendor-independent, so CMSIS-NN should work for all Arm Cortex-M processors. You will get the most performance benefits if you use a Cortex-M core with DSP extensions, like M4, M7 or M33. 

IMHO, the choice of development boards/platforms really comes down to your preferred embedded development environments and specific board features (e.g., sensors, I/O, network connectivity, memory size etc.) for your application. Based on my experience, lack of supported drivers is usually the biggest problem.

If you prefer visual studio/eclipse like IDE, you can take a look at [KEIL](http://www.keil.com/dd2/) which supports a wide range of boards. If you prefer either online IDE or command-line interface, you can take a look at [mbed](https://os.mbed.com/platforms/) compatible ones with either the online IDE or [mbed-cli](https://github.com/ARMmbed/mbed-cli) command-line tools. There are also vendor-specific tools which should support their boards (like the above mentioned OpenMV) decently. But I don't have much experience with those.

We have been using many different boards. For example, we have released some [examples](https://github.com/ARM-software/ML-KWS-for-MCU/tree/master/Deployment) using mbed with [DISCO_F746NG ](https://os.mbed.com/platforms/ST-Discovery-F746NG/)and [FRDM-K64F](https://os.mbed.com/platforms/FRDM-K64F/). 

So short answer is: 
* We don't expect any compatibility issue for using CMSIS. You should use M4/M7/M33 for performance 
* pick the board based on the preferred tool-chain and board features that match your application

I hope this answers your question. Let me know if you have any more questions.

Thanks,
Liangzhen

uu-kameshm, 2018-06-26:
Hi Liangzhen

Thank you, for the detailed inputs.

Could you let me know, on which h/w reference boards/platforms the CMSIS NN based applications have been successfully tested (with Linux or similar OS). Also, it would help to know which networks are already tested (such as AlexNet etc., if known).

This is just to make sure, I do not get into any unforeseen issues with respect to tool chain etc. and avoid debugging.

Thanks
Kamesh