ARM-software/CMSIS_5 #385: RTX Cortex-M header file does not protect against redefinition for STRINGIFY

0xc0170, 2018-06-28:
See https://github.com/ARMmbed/mbed-os/pull/7329. 

File in question : `rtx_core_cm.h`. There are some macros protected against redefinition:

```
#ifndef DOMAIN_NS
#define DOMAIN_NS               0
#endif
```

Helper like this one `#define STRINGIFY(a) #a` is not. I checked the rest of defines, they are RTX specific so should be fine. This is causing lot of warnings as shown in the link above (each file pulls in this header file). 
I would suggest either this helper to be renamed or check if its not already defined (expecting stringify defined to be the same).

Please review.

JonatanAntoni, 2018-06-28:
`rtx_core_cm.h` is internal and should not be included from outside. Hence we don't care that much about using exclusive names for our internal helper macros. Please try to get rid of including this file.

0xc0170, 2018-06-28:
👍  I'll close this issue and fix our inclusion problem.

closed 2018-06-28 by 0xc0170