ARM-software/CMSIS_5 #386: failing to initialize oskernelInitialize() correctly

zawbaz, 2018-07-01:
I am using CMSIS RTX RTOS2 libraries on apollo2mcu cortex-m4f in DS5. The return value of osKernelInitialize() appears as osError. When the function SVC0_0 (KernelInitialize,           osStatus_t)  is called inside rtx_kernel.c , the link register gets the value of some reserved region and SVC_Handler is called and osKernelInitialze() returns always osError. Any help would be highly appreciated in this regard

JonatanAntoni, 2018-07-02:
Hi @zawbaz,

thanks for getting in touch. The misbehavior you describe sounds like a fundamental issue to me.
- What "value of some reserved region" do you refer to? On Cortex-M devices the link register contains a magic value like `0xFFFF FFFn` when jumping to handler mode. This value does not actually address a reserved memory region but encodes certain system states like the ones hold in `CONTROL` register (stack selection and privilege mode bits).
- Have you tried to run one of our Blinky examples on your target? E.g. the CMSIS pack contains an example `CMSIS-RTOS2 Blinky (uVision Simulator)`. This example should run on your target with minimal changes, e.g. some low level initialization for clocks and interrupts.
- Can you share a stripped down reproducer for your issue? I.e. if `osKernelInitialze()` fails the low level initialization might be the cause.

Cheers,
Jonatan

zawbaz, 2018-07-02:
Hi Jonathan, 
                  Thanks a lot for your reply. Yes the link register contains 0xFFFF FFE9, i thought this is the wrong value . But you clarified it that it is the magic value . Yesterday I tried with the optimization level Restricted (-01) on DS5 then the kernel is Initialized and thread is also created successfully . But on optimization level Minimum (-O0)  , osKernelInitialze() is gving status of osError . 
With the  optimization level Restricted (-01), osKernelInitialze() , osThreadNew(app_main, NULL, NULL) and  osKernelStart() are working fine but sometimes the gdb shows No source available for "(gdb[59].proc[42000].threadGroup[i1],gdb[59].proc[42000].OSthread[1]).thread[1].frame[0]" 
or No source available for "0x0"        
I have not tried CMSIS-RTOS2 Blinky (uVision Simulator) because i am using DS5. 

Regards,
Faisal

JonatanAntoni, 2018-07-02:
Hi @zawbaz,
can you provide a stripped-down minimal example reproducing your issue, please?
There are many details that could cause such issues. Hard to tell without knowing them all, e.g. compiler version, compiler and linker flags, interrupt and rtos configuration, aso.

Cheers,
Jonatan

zawbaz, 2018-07-03:
Dear Jonathan, 
                     I am using ARM Compiler 5 (DS-5 builtin) and the default settings of RTX_Config.h of RTX RTOS2. main.c template file  of CMSIS-RTOS2 in Apollo2mcu has been used. The isr names in startup.s  file of apollo2 mcu have been modified according to irq_cm4f.s. 
This is the compiler options --cpu=Cortex-M4.fp.sp --apcs=/hardfp --thumb --littleend -D_RTE_ -DAPOLLO2_1024
Debugging is being performed through GDB-Segger J-Link Debugging in DS5 
I am facing no issue in Keil Uvision. If you require other information,please let me know 

Regards, 
Faisal

JonatanAntoni, 2018-07-04:
Hi Faisal,

Now its getting strange. Did you raise this issue at the DS-5 support, too?

You could try to compare the MDK compiler options with the ones in DS-5. What happens if you use the binary from MDK and debug it with DS-5? And vice versa?

Can you provide the projects?

Cheers,
Jonatan

zawbaz, 2018-07-04:
Hi Jonatan,  
               Yes I will ask DS-5 support too. But I think it can be the problem of GDB client or server or GNU Mcu eclipse plug in . Because i m using the gnu mcu eclipse plugin for jlink debugger. I will try also other method as you suggested . 

Regards, 
Faisal

JonatanAntoni, 2019-03-12:
@zawbaz, anything new about this? Has your issue been solved in the meanwhile? Consider providing further information or closing the issue, please.

closed 2019-07-30 by JonatanAntoni