ARM-software/CMSIS_5 #387: User Poll for CMSIS-Pack: Cclass names for standardization - please submit proposals

ReinhardKeil, 2018-07-06:
CMSIS-Pack is a technology that allows to distribute software components.  Currently there is a set of Cclasses are defined for arranging software components (see http://arm-software.github.io/CMSIS_5/Pack/html/element_taxonomy.html#element_taxonomyDescription - Table: Component Classes)

We got feedback that we should further define/standardize Cclass names.  In recent Cloud connector packs (available here: https://github.com/mdk-packs) we used: 

- Cclass=“Data Exchange” (Cgroup=<protocol>)     // for a JSON component
- Cclass=“IoT Client” (Cgroup=<cloud vendor>)      // for various Cloud clients
- Cclass=“IoT Utility” (Cgroup=“MQTTPacket”, “Socket (API)”,   // for utility interfaces 
- Cclass=“Security” (Cgroup=“mbed TLS”)               // for encryption

I would like to ask the user community to give feedback in this issue.  Should we further standardize and define Cclass names for software components?

iomint, 2018-10-10:
It sounds indeed useful to define additional Cclass names to cover new domains as the ones you propose. Doing it in a coordinated way with a review of the proposals by the community sounds a good idea as it could avoid unnecessary fragmentation and allow to keep a centralized location for the standardized Cclass names.
Further to your proposals, find below additional  ones:
•	Class names can be advantageously used to structure the source code tree so the first proposal is to add to the existing “Board Support” Cclass two new Cclass names for this purpose.
The first one aims to identify drivers for (hardware) components available on development boards aside the MCU itself. Cgroup can be used to identify the functions or combination of function.  
        o  Class =”Board Component” (Cgroup=”LCDController”, “IOExpander”, “GyroMag”, …) 
The second one aims to address the drivers needed for an extension boards (aka Shield) 
        o  Class=”Board Extension” (Cgroup=)
•	The third new Cclass name proposal aims to create a Cclass for utility (sw) components
        o  Class=”Utilities” (Cgroup=”log”, “fonts”, …)
•	The fourth proposal aims to define a Cclass for audio processing (sw) components
        o   Class=”Audio” (Cgroup=”encoder”, “decoder”, “equalizer”, …)

jkrech, 2018-10-10:
All your extensions make sense to me. 

The term "Component" in "Board Component" may get people confused as these are all software components. We could clarify by using "Board HW Component" (HW vs. SW should be widely used abbreviations) or use the word "Board Part" instead.

For "Board Extension" would it make sense to rather name it "Extension Board" as I am expecting the Cgroup to specify the name of an extension board or shield. Am I getting the context wrong?

I would propose using "Utility" in singular for consistency with the other class names.

**- Cclass="Board Part"
- Cclass="Extension Board"
- Cclass="Utility"
- Cclass="Audio"**

loverdeg-ep, 2018-12-30:
@ilg-ul

ReinhardKeil, 2019-03-13:
Added several new Cclass names for components.  See https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/DoxyGen/Pack/src/pack_creation.txt#L81

loverdeg-ep, 2019-03-13:
@theotherjimmy Is this relevant to mbed-os at all?

theotherjimmy, 2019-03-13:
Thanks for the heads up @loverdeg-ep. At this time this is not relevant.

@ReinhardKeil I was under the impression that while the some of the `Cclass`es are standard, the current tools allow any `Cclass` to be used. So long as we continue to allow this experimentation, I think the proposed standardization (stabilization?) sounds good. I'm a little concerned with the use of "IoT" in the names, as I generally don't associate "IoT" with any sort of software and instead associate "IoT" with a particular class of hardware and software techniques or an industry buzz-word. I don't have a good alternative though. Based on your comments, it seems like "IoT Client" could just as easily be called "Cloud Client" and "IoT Utility" could use some more defining features before standardization the name.

Perhaps it would be productive to discuss each standardization separately?

ReinhardKeil, 2019-03-13:
Jimmy you are correct, the usage of these Cclass names are just recommendations. We are using the IoT class names right now for your IoT connectors. While Cloud Connector would work too, we decided to use IoT as this is the name that seems to resonate with users. Let’s see what other think here once we released it. There is always room for reasonable change requests.

Reinhard

IMPORTANT NOTICE: The contents of this email and any attachments are confidential and may also be privileged. If you are not the intended recipient, please notify the sender immediately and do not disclose the contents to any other person, use it for any purpose, or store or copy the information in any medium. Thank you.

theotherjimmy, 2019-03-13:
@ReinhardKeil Thanks for the clarification. I'm satisfied with the direction you're taking.

ReinhardKeil, 2019-03-20:
Closing this issue as it is completed

closed 2019-03-20 by ReinhardKeil