ARM-software/CMSIS_5 #391: arm_fir_f32 implementation

juliobc, 2018-07-12:
Hi, I have a problem with the result of  arm_fir_f32(). I made a copy of the next example:

http://www.keil.com/pack/doc/CMSIS/DSP/html/arm_fir_example_f32_8c-example.html

I have the p_result with the filtered signal, but the testOutput is displaced NUM_TAPS/2 values and I lost the end of the input signal.

Is this functionality correctly?
I show the code and images:

`
	#define NUM_TAPS	3001
	uint32_t signal_length = 4096;
	float32_t* p_signal = (float32_t*)&sram_raw_signal_address[0];
	float32_t* p_filter = (float32_t*)&sram_raw_signal_address[16384];
	float32_t* p_result = (float32_t*)&sram_raw_signal_address[32768];
	
	for (uint16_t i=0; i<NUM_TAPS; i++)
		p_filter[NUM_TAPS - 1 - i] = NUM_FIR_COEF[i];
		
	/* Create input signal: */
	uint32_t f_hz = 50000;
	float32_t w =2*PI*f_hz;
	for (uint16_t i=0; i<1000; i++)
		p_signal[i] = 2.6f*(sinf(w*i/Fs) + 0.24f*sinf(2.1f*w*i/Fs));
	for (uint16_t i=1000; i<signal_length; i++)
		p_signal[i] = sinf(w*i/Fs) + 0.24f*sinf(2.1f*w*i/Fs);
	
	#define BLOCK_SIZE	64
	float32_t firStateF32[BLOCK_SIZE + NUM_TAPS - 1];
	arm_fir_instance_f32 S;
	arm_fir_init_f32(&S, NUM_TAPS, p_filter, firStateF32, BLOCK_SIZE);
	uint32_t numBlocks = signal_length/BLOCK_SIZE;
	for(uint32_t i=0; i < numBlocks; i++)
		arm_fir_f32(&S, p_signal + (i * BLOCK_SIZE), p_result + (i * BLOCK_SIZE), BLOCK_SIZE);
	`

[image]

[image]

[image]

llefaucheur, 2018-07-12:
Hi Juliobc, the FIR coefficient are symetrical and real, the filter has a linear phase. Consequently, the group-delay is equal to "number of taps"/2. The filtering function starts with memory state with null-samples and create an output having the same size of the input, the last samples seems "lost", but they are inside the state memory of the FIR. If you add more zeroes at the end of the input buffer, you will recover the missing samples. See also https://dspguru.com/dsp/faqs/fir/properties/

juliobc, 2018-07-12:
Hi @llefaucheur 

ok, then I have to add more zeroes to the end of my input. 
Thank you

closed 2018-07-12 by juliobc