ARM-software/CMSIS_5 #393: FFT implementation with greater buffer

juliobc, 2018-07-13:
Hi, I need to make a FFT and a iFFT with a buffer of 8192 points or more because I am trying to make a fast convolution with a filter of a lot of points. I have a huge RAM memory.

Do you think that is possible in a future with CMSIS DSP library??

Thank you

llefaucheur, 2018-07-13:
We have a continuous process of improvement of CMSIS-DSP for new features and performance improvements. Doing FFT of up to 64k points is in our plans. We have the action to rank the tasks. 
What is your focus ? I suppose Float32. Regards, Laurent.

juliobc, 2018-07-13:
Hi @llefaucheur 
yes, I would like work with float32

uTasker, 2018-09-17:
Any news on 8192 point FFT?
I have tried extending it but can't work out how to generate the bit reversal table at the end.
Generating a radix 8 table (as used by 4096) is no problem but what is needed in the (8x2) case????

kevingoh, 2019-05-21:
I am also on the same quest ( making the library for longer FFT ). I have managed to compute Real FFT for 8192, by using existing tables for CFFT, and generating my own TwiddleFactor Table for 8192 points using Matlab.
However, I still cant figure out how to generate the bit reversal index table yet. 
I have found traces of the cmsis library to a book by Burrus, which may provide further clue, but unfortunately I do not have the time to fully digest the book yet.

The link to the book is [here](https://eng.libretexts.org/Bookshelves/Electrical_Engineering/Book%3A_Fast_Fourier_Transforms_(Burrus))

uTasker, 2019-05-21:
It is almost a year now that this extension has been waited for. We have all been able to work out the twiddle factors and have firmware just waiting for the bit reversal table, to which the secret is presumably held by the original CMSIS DSP FFT developer and no one else has been able to work out where it comes from.
Is there a chance that the algorithm to generate it could be disclosed if the table itself can be provided?

kevingoh, 2019-05-21:
No I don't think it is that secretive, but rather it looks like someone critical to the initial development of the conversion (I think the algorithm was translated from a FORTRAN program from Burrus of RICE University) has stopped collaborating in this project.
I have strong confidence that the bit reversing or unscrambler method can be recovered very soon.

kevingoh, 2019-05-24:
Hi, I think I have a clue how the bitreversal tables are generated for the mixed radix cases, as described in the patent US5473556A.

General steps:
1. Generate indices from 0 to len-1
2. turn them into binary
3. encode them with stage specific wide bit-fields
4. flip the bit-fields around
5. run through final round of optimization to remove redundancy.

I have managed to generate an almost matching list using Octave, as I am now working on the 5th step above.

Please help to verify my finding.

Happy weekends!

die4lie, 2019-10-25:
Hi,there. Are there any progress?? Also I have question about non 2^N array sizes. Is it possible to use for example 4092 or 1985 or something like that? Are there any methods to divide iterations with different 2^N sizes or something like that?

die4lie, 2019-10-25:
Seems today is the day when patent expires:

"Application US08/117,959 events
1992-04-30
Priority to US87794992A
1993-09-08
Application filed by Sharp Corp, Sharp Microelectronics Technology Inc
1993-09-08
Priority to US08/117,959
1995-12-05
Application granted
1995-12-05
Publication of US5473556A
1999-04-02
Assigned to SHARP ELECTRONICS CORPORATION
2012-12-05
Anticipated expiration
2019-10-25
Application status is Expired - Lifetime"

uTasker, 2019-10-25:
15 months and counting. Hopefully no one needs any real urgent support.....

christophe0606, 2020-03-11:
I have created this issue : https://github.com/ARM-software/CMSIS_5/issues/858
It explains how to generate the bit reversal tables and contain the bit reversal table for a 8192 FFT.

The 8192 case won't be included by default in the CMSIS-DSP because the binary pack is already getting too big with all the tables. Also the 8192 version would require us to change the type of the bit reverse table to contain uint32_t instead of uint16_t (because the tables are expressing the offsets in bytes).

And we'd like to avoid changes in the public API.

closed 2020-03-26 by JonatanAntoni

Alex-EEE, 2021-04-27:
Just a note for anyone coming across this, you can create a large FFT from smaller ones using the "overlap-add" or "overlap-save" method.  Just google it.

Treeed, 2023-07-26:
I have combined some of the available code and information and written some python code to generate an extension to the CMSIS DSP with longer FFTs. Maybe this is useful to some: https://github.com/Treeed/Long_FFTs_for_CMSIS_DSP/tree/master

christophe0606, 2023-07-27:
@Treeed Thank you. 

CMSIS-DSP now has its own repository. And the wiki contains some explanations about how to generate table for longer FFTs (and a Python script provided by community that you link in your post).

https://github.com/ARM-software/CMSIS-DSP/wiki/Generating-Bit-Reverse-Tables-for-FFTs

If you think it may be interesting to have a compilation directive to select uint16 or uint32 for the armBitRevIndexTables , you can open an issue on the new repository.