ARM-software/CMSIS_5 #395: cast-align warning

pavelrevak, 2018-07-18:
I getting this warning with casting

```
    /CMSIS/DSP/Include/arm_math.h:4957:35: error: cast from 'q15_t* {aka short int*}' to 'int32_t* {aka long int*}' increases required alignment of target type [-Werror=cast-align]
         vstate = __SIMD32_CONST(S->state);
                                   ^
    cc1plus.exe: all warnings being treated as errors
```

- gcc v6.3.1
- cortex-m7
- `g++ -Werror -Wcast-align ....`

llefaucheur, 2018-07-18:
Hi Pavel, is it possible to disable the gcc warning (I am not familiar with gccv6).
Changing the code would represent a substantial effort. One other option we are currently considering is to port the code in assembly to avoid the dependency to the compilers.
Regards, Laurent

llefaucheur, 2018-07-19:
Hi Pavel, I have a question  for you.
Why do you need to recompile the library since it is available in binary form ? In other words: did you modified the code for your needs ? Regards, Laurent

pavelrevak, 2018-07-19:
Hi llefaucheur, thanks for quick response.
1. We building everything from source.
2. But after analysis seems that the array of 16 bit words `state`  is not aligned to 32 bit and the conversion into 32 bit is making the warning.

Larry7, 2019-04-03:
We also had this problem

Larry7, 2019-04-03:
arm_math.h: In function 'q15_t arm_pid_q15(arm_pid_instance_q15*, q15_t)':
arm_math.h:471:54: error: cast from 'q15_t* {aka short int*}' to 'int32_t* {aka int*}' increases required alignment of target type [-Werror=cast-align]
 #define __SIMD32_CONST(addr)  ((__SIMD32_TYPE *)(addr))
                                                      ^
arm_math.h:4957:14: note: in expansion of macro '__SIMD32_CONST'
     vstate = __SIMD32_CONST(S->state);
              ^~~~~~~~~~~~~~

Larry7, 2019-04-04:
I tried to modify arm_math.h
from
```
#define __SIMD32_CONST(addr)  ((__SIMD32_TYPE *)(addr))
to
#define __SIMD32_CONST(addr)  ((__SIMD32_TYPE *)((void*)addr))
```

JonatanAntoni, 2019-04-04:
Hi @Larry7,

This macro breaks the strict aliasing rule by casting pointers to different types.
While the macro is still in the header for backward compatibility reasons we removed the usage of that macro from CMSIS-DSP algorithms with the last release.

Please check CMSIS-DSP 1.6.0 shipped with CMSIS 5.5.1.

Cheers,
Jonatan

JonatanAntoni, 2019-05-21:
Closing due to no further activity. Feel free to reopen if any open concerns left to be discussed.

closed 2019-05-21 by JonatanAntoni