ARM-software/CMSIS_5 #396: SVD Output decoupling macro in IO header.file

ponsuke606, 2018-07-24:
Hi
I'm try making SVD and IO header file.
I could generate IO header file. But, there are decoupling macro in it.
If it is possible , I would like to solve this decoupling issue.

E.G.(Decoupling definitions)
/* ======================  ELSR  ============================== */
#define ELC_ELSR_ELS_Pos                  (0UL)                     /*!< ELC ELSR: ELS (Bit 0)                          */
#define ELC_ELSR_ELS_Msk                  (0xffffUL)                /*!< ELC ELSR: ELS (Bitfield-Mask: 0xffff)    */
/* ======================  ELSR  =============================== */
#define ELC_ELSR_ELS_Pos                  (0UL)                     /*!< ELC ELSR: ELS (Bit 0)                 */
#define ELC_ELSR_ELS_Msk                  (0xffffUL)                /*!< ELC ELSR: ELS (Bitfield-Mask: 0xffff)           */

I wrote registers name using "%s" definition in SVD file. 

<---About  ELSR14 and ELSR15-->
               <dimIndex>14,15</dimIndex>
               <name>ELSR%s</name>
                       == omit ==
<---About  ELSR18 and ELSR19-->
                 <dimIndex>18,19</dimIndex>
                  <name>ELSR%s</name>
                          == omit ==

This sample file is following.
[test.zip](https://github.com/ARM-software/CMSIS_5/files/2222252/test.zip)

I think this is occurred by multiple definitions of  "%s" for register name.
Do you have any solutions?

Thanks.

JonatanAntoni, 2019-03-15:
Hi @ponsuke606, 

May I ask you to check the latest version of SVDConv for your scenario, please?
You should now get the _Pos and _Msk defines once per register, i.e.
```
/* ====================== ELSR ============================== /
#define ELC_ELSR14_ELS_Pos (0UL) /!< ELC ELSR: ELS (Bit 0) /
#define ELC_ELSR14_ELS_Msk (0xffffUL) /!< ELC ELSR: ELS (Bitfield-Mask: 0xffff) /
/ ====================== ELSR =============================== /
#define ELC_ELSR15_ELS_Pos (0UL) /!< ELC ELSR: ELS (Bit 0) /
#define ELC_ELSR15_ELS_Msk (0xffffUL) /!< ELC ELSR: ELS (Bitfield-Mask: 0xffff) */
```

The problem we face here is, that using %s at register level does not create a C-type array. Even if the registers are named similar their bit fields might differ, i.e. ELSR14/15 and ELSR18/19 in your example are declared independently.

ponsuke606, 2019-04-09:
Hi @JonatanAntoni ,

Thank you for fixed issue!
I have checked it and it  worked properly!

[image]

left side:Previous version of SVDConv.exe
right side: latest version of SVDConv.exe

JonatanAntoni, 2019-04-09:
@ponsuke606, close the issue if you think its solved, please.

closed 2019-04-09 by ponsuke606