ARM-software/CMSIS_5 #397: How to realize batch norm layer in CMSIS NN?

hexenzhou, 2018-07-30:
Or whether the BN layer has started quantifying after the weights has been calculated?

dtch1997, 2020-07-06:
Has this issue been resolved? I would like to deploy MobileNet in CMSIS-NN and batch norm is required.

felix-johnny, 2020-07-06:
@d-kato No, batch normalization isn't supported and is not in the immediate plans. The work done now is on TFLu compliant functions (https://github.com/ARM-software/CMSIS_5/tree/develop/CMSIS/NN#legacy-vs-tfl-micro-compliant-apis) and there you would use TFLu as the fallback for anything not supported in CMSIS-NN.

d-kato, 2020-07-06:
@dtch1997  Probably not a message to me, but a message to you.  
@felix-johnny  That's right?

felix-johnny, 2020-07-06:
yes!

dtch1997, 2020-07-08:
@felix-johnny thanks for the quick response. To be clear, you are suggesting that I can implement BatchNormalization using the TFLu compliant APIs? Specifically, elementwise add and elementwise multiply.

felix-johnny, 2020-07-09:
@dtch1997 Sorry, I do not know how your setup looks like or what framework you are using. All I am saying is, if you use TFLu as your framework you and build it for cmsis-nn(E.g https://github.com/tensorflow/tensorflow/tree/master/tensorflow/lite/micro/kernels/cmsis-nn) you get a list of optimized kernels that you see in the link above. As for the rest, if it is available in TFLu(Check https://github.com/tensorflow/tensorflow/blob/master/tensorflow/lite/micro/all_ops_resolver.cc) you are good to go.