ARM-software/CMSIS_5 #398: layer type for flatten and dropout

kidswong999, 2018-07-31:
Hi, I found that cmsis-nn just support 'data','convolution','relu', 'pooling', 'innerproduct' layers.

My mnist example include  'flatten' layer (reshape data), maybe ‘dropout’ layer is also useful.

Is any plan to add this common layers?

JonatanAntoni, 2018-07-31:
Hi @hexenzhou, @kidswong999,

I expect the functions available will grow over time. CMSIS-NN is an open source project and hence community contributions are highly appreciated. Please feel free to provide additional functionality you consider valuable by raising pull requests. 

Thank you for your collaboration,
Jonatan

hexenzhou, 2018-08-01:
Hi @JonatanAntoni

Thank you for your reply, I am looking forward to adding the Batch Normalization layer in CMSIS-NN, which is very important in my CNN model.

Thanks again to the arm team for open source, this project has helped me a lot.

BabaJaguska, 2019-04-11:
Indeed. Unless you want to simulate some sort of uncertainty, but I'm not sure why you'd want to do that on a chip. The original dropout [paper](http://jmlr.org/papers/volume15/srivastava14a.old/srivastava14a.pdf), however, mentions inference scaling of the weights connecting the dropout layer with the keep probability of the dropout used during training. Then again, Keras just turns off dropout at test time.

felix-johnny, 2021-01-07:
@kidswong999  CMSIS-NN is mainly for optimization of compute intense operators like convolution or SVDF. For memory manipulation operators and lesser used operators, it relies on the reference implementation by TensorFlow Lite micro. We do have reshape but, it doesn't add any value in terms of optimization. There are no plans to add support for Dropout.

closed 2021-01-07 by felix-johnny