ARM-software/CMSIS_5 #401: SVDConv: New feature dsppresent

simtind, 2018-08-10:
It would be useful to be able to declare the __DSP_PRESENT macro from the SVD file. 
I propose a new tag <dspPresent> working similar to <fpuPresent>, either mandatory, or default not present.

ReinhardKeil, 2018-08-17:
What should __DSP_PRESENT express?  The availability of SIMD instructions?
Every Cortex-M4/M7 has this extensions, therefore I think it is redundant.

simtind, 2018-08-17:
DSP / SIMD instructions are optional in Cortex-M33. Referencing https://developer.arm.com/products/processors/cortex-m/cortex-m33 .

From CMSIS-DSP documentation:
http://www.keil.com/pack/doc/CMSIS/DSP/html/index.html 
__DSP_PRESENT:
Initialize macro __DSP_PRESENT = 1 when Armv8-M Mainline core supports DSP instructions.

ReinhardKeil, 2018-08-17:
Thanks for clarifying.  Yes Cortex-M33 has optional SIMD instruction set.

JonatanAntoni, 2019-03-12:
@jkrech: We added __DSP_PRESENT and __DSP_USED to ARMCM33 device files. What about adding this to SVDConv?