ARM-software/CMSIS_5 #403: [enhancement] Independent ../DSP/Source/CommonTables/arm_const_structs and arm_common_tables

fgr1986, 2018-08-13:
Hi, 
A great enhancement would be hierarchical common tables `CMSIS_5/CMSIS/DSP/Source/CommonTables/arm_const_structs.c` and `CMSIS_5/CMSIS/DSP/Source/CommonTables/arm_common_tables.c`
This would reduce considerably the size of the generated binary if only one function is required, for example, an FFT of fixed length.
When memory is a concern having independent hierarchical tables helps to reduce the overall ROM size.

Thank you

llefaucheur, 2018-09-03:
Hi frg1986, would you be ok with a change of the legacy API ?
This will translate in a change in the initialization phase, where you will select the right table with a dedicated function. Regards, Laurent

fgr1986, 2018-09-04:
@llefaucheur Thank you. Having dedicated functions (such as FFT512_qX, FFT128_qX, etc) only calling the required constants would be perfect.

christophe0606, 2019-05-29:
We have reorganized a little bit the tables (for rfft) and added compilation flags. Now it is possible to build CMSIS-DSP with only the needed tables. It applies to FFT but also sin/cos and lms.

closed 2019-05-29 by christophe0606