ARM-software/CMSIS_5 #405: Startup init order

KarolisMil, 2018-08-16:
In CMSIS/Device folder, GCC toolchain files:
why SystemInit is called after static data initialization (copying, setting) and not before? Every argument that comes to my mind says that this can be improved. I would love to know the reasons. Thank you!

ReinhardKeil, 2018-08-17:
I agree this is strange.  The issue might be that clearing memory also deletes stack content. We need to verify that.

KarolisMil, 2018-08-17:
After SystemInit, system could be more versatile. One of many examples migth be with clocks configured in SystemInit. That means data copy and setting ZI might be done faster if so. But this is just as example and it is up to user what to do there in SystemInit. The question is wheather he/she has that oportunity or needs to change official example files before use...

salkinium, 2018-08-23:
`SystemInit` may be written in C and may therefore use all of C’s features like reading a initialized `.data` variable in SRAM. If `SystemInit` is called before the `.data` or `.bss` sections have been copied/initialized, you may read random, or after a software restart, even old values.
So unless you write `SystemInit` in Assembly, this is a bit of a 🐔 and 🥚 problem.

ilg-ul, 2018-08-23:
@salkinium, this sounds like a bit of a 🐔 and 🥚 problem, but you can also write `SystemInit()` in assembly and, if careless, access external variables that were not initialised; the same as you can write it in C and, if careful, avoid such variables. The language you use to write this function is irelevant, only it's content is.

However it is true that calling `SystemInit()` before `.data` or `.bss` sections have been copied/initialized has the disadvantage that it cannot leave any results in global variables, because they will be erased shortly.

Actually, for Cortex-M, the entire startup sequence can be written in C, the only real danger is that without some cryptic options, modern compilers may not like pointers initialised from linker scripts, and trigger undefined behaviour.

salkinium, 2018-08-23:
> the same as you can write it in C and, if careful, avoid such variables.
> Actually, for Cortex-M, the entire startup sequence can be written in C

Yeah, at least for GCC that works fine, since it behaves like you'd expect.
Don't ask a compiler engineer about this though, they'll tell you all about the C Abstract Machine and UB until your ears fall off. That's why all the startup scripts are still in asm even though it's marketed as "C compatible". #AtLeastTheStackPointerIsSetInHardware

> without some cryptic options, modern compilers may not like pointers initialised from linker scripts

Examples? I've discovered the [requirement to use `--whole-archive` for linking against static libs](https://github.com/embeddedartistry/meson-buildsystem/issues/5) containing custom linker sections.

KarolisMil, 2018-08-23:
@salkinium, I agree, but not completely. Take a look at ARMCC toolchain example. Now GCC is opposite. I remember it being like that for ages. ARMCC startup example calls SystemInit before __main entry point. It is usual for SystemInit no to use nor access any global data which is not initialized yet. It's purpose to initialize critical system things, not the whole HAL lib etc. For example, enable FPU. I am coming from ARMCC world and this seems not sexy.

David-Garcia-Polo, 2018-08-24:
Please note that in Cortex-M SystemInit() at least global variable SystemCoreClock needs to be updated, at least according to the templates. You cannot do that before initialization. SystemInit() is correct where it is.

Now, since SystemCoreClock is not used by the debuggers yet as far as I know, the variable could be eliminated, but that is another issue.

ilg-ul, 2018-08-26:
If `SystemInit()` is called before bss init, to update `SystemCoreClock`, you must call `SystemCoreClockUpdate()`.

JonatanAntoni, 2018-09-03:
Hi all,

Bad luck :)

This inconsistency between GCC and Arm/IAR Compilers is in the template since ages and nobody realized that, yet.

I think the reason is because Arm and IAR Compilers handle .bss and .data sections automatically during library startup, i.e. simply jumping to `__main` or `__iar_program_start`. `SystemInit` is always called at the very beginning of the `ResetHandler` and hence before global data is initialized. 

With GCC we have a slightly different behavior because .bss/.data sections are handled manually within the `ResetHandler` and (unfortunately) before executing `SystemInit`. 

Fortunately all the startup code is only a template intended to be changed according to the users needs. So please feel free to move the call to `SystemInit` up to the very beginning of `ResetHandler` to get the same behavior as with other compilers.

I recognize this as a bug in the startup templates we should fix, i.e. call `SystemInit` at the very beginning by default. Does this makes sense to you?

Thanks,
Jonatan

KarolisMil, 2018-09-04:
Totally agree!

JonatanAntoni, 2019-03-11:
@KarolisMil, may I ask you to cross check the fix provided and close the issue, please?

JonatanAntoni, 2019-05-21:
Closing due to no further activity. Please feel free to reopen if any open concerns left.

closed 2019-05-21 by JonatanAntoni