ARM-software/CMSIS_5 #413: Undefined __CMSIS_GCC_OUT_REG and __CMSIS_GCC_USE_REG under Core A

Beetix, 2018-08-23:
Hi,

I have noticed that defines `__CMSIS_GCC_OUT_REG` and `__CMSIS_GCC_USE_REG` are missing in cmsis_gcc.h under Core A. These defines are used in functions `uint32_t __REV(uint32_t value)` and `int16_t __REVSH(int16_t value)`. Is there a reason for not having the same defines as in cmsis_gcc.h under Core ?

Thank you

JonatanAntoni, 2018-09-03:
Hi @Beetix,

This looks like a copy'n'paste error that got through because we don't actively support the GCC versions in question. For Cortex-M we need to distinguish between Thumb and Thumb-2 modes (because of Cortex-M0). On the other hand Cortex-A is always Thumb-2 so we don't need this here.

We should clean up the header file and substitute the macros with static "r" or "=r" strings. A pull-request for that is very appreciated.

Thanks,
Jonatan

Beetix, 2018-09-03:
Hi @JonatanAntoni,

Ok, thank you for this clarification. I'll be happy to make a pull-request.

Benjamin

PR #432 (2018-09-20): Fixed issue #413
- Fixed #413 copy'n'paste error from Core to Core A
closed 2018-09-20 by JonatanAntoni