ARM-software/CMSIS_5 #414: MMU section and page creation functions always make physical address = virtual address

Beetix, 2018-08-23:
Hi,

I have noticed that the MMU section and page creation functions always make physical address = virtual address. For example, here is the current function to create a section from core_ca.h:
```c
/** \brief  Create a 1MB Section
  \param [in]               ttb  Translation table base address
  \param [in]      base_address  Section base address
  \param [in]             count  Number of sections to create
  \param [in]     descriptor_l1  L1 descriptor (region attributes)
*/
__STATIC_INLINE void MMU_TTSection(uint32_t *ttb, uint32_t base_address, uint32_t count, uint32_t descriptor_l1)
{
  uint32_t offset;
  uint32_t entry;
  uint32_t i;

  offset = base_address >> 20;
  entry  = (base_address & 0xFFF00000) | descriptor_l1;

  //4 bytes aligned
  ttb = ttb + offset;

  for (i = 0; i < count; i++ )
  {
    //4 bytes aligned
    *ttb++ = entry;
    entry += OFFSET_1M;
  }
}
```

Here the section index is calculated using the base address (`offset = base_address >> 20`). Then the section physical address is calculated once again from the base address (`base_address & 0xFFF00000`). Consequently we get physical address = virtual address.

Wouldn't it be better, in order to have the freedom to map virtual addresses to any physical addresses, to have an implementation like this:
```c
/** \brief  Create a 1MB Section
  \param [in]                   ttb  Translation table base address
  \param [in]  virtual_base_address  Section virtual base address
  \param [in] physical_base_address  Section physical base address
  \param [in]                 count  Number of sections to create
  \param [in]         descriptor_l1  L1 descriptor (region attributes)
*/
__STATIC_INLINE void MMU_TTSection(uint32_t *ttb, uint32_t virtual_base_address, uint32_t physical_base_address, uint32_t count, uint32_t descriptor_l1)
{
  uint32_t offset;
  uint32_t entry;
  uint32_t i;

  offset = virtual_base_address >> 20;
  entry  = (physical_base_address & 0xFFF00000) | descriptor_l1;

  //4 bytes aligned
  ttb = ttb + offset;

  for (i = 0; i < count; i++ )
  {
    //4 bytes aligned
    *ttb++ = entry;
    entry += OFFSET_1M;
  }
}
```
The same would apply for page creation.

Thank you

JonatanAntoni, 2018-09-03:
Hi @Beetix,

You can of course use the MMU to "hide" physical addresses from the application. I'd like to understand the use case you have in mind to do so? Typical bare metal firmware I have seen so far does not incorporate complex memory models.

Cheers,
Jonatan

Beetix, 2018-09-03:
Hi @JonatanAntoni,

It seems odd to me since one of the main purposes of the MMU is to provide an abstraction of the physical memory. I agree that most bare metal firmwares might not need this. I think it could be useful in a context where you would want to run multiple applications with their own address space orchestrated by a custom OS. My concern is mainly to leave the possibility open. Plus, it doesn't seem that many changes are needed to allow this.

Thank you for your answer,

Benjamin