ARM-software/CMSIS_5 #415: IAR compilation problem with __RESTRICT define

c1728p9, 2018-08-25:
Building for IAR fails due to the `__restrict` keyword not being recognized. When `__RESTRICT` is defined to `restrict` builds work normally. This change was introduced in the commit 2fd08630a4ba98673ae5e2830a1689a59f99144a so 5.4.0 is the first release which contains this change.

For reference I found this issue when [upgrading mbed-os to version 5.4.0](https://github.com/ARMmbed/mbed-os/pull/7875).

TTornblom, 2018-09-03:
Looks like "__restrict" is only recognized in EWARM 8.x, while EWARM 7.x only recognizes "restrict", which needs the "--enable_restrict" compile option.

So this change needs rework to work with EWARM 7.

JonatanAntoni, 2018-09-03:
May I ask you both (@TTornblom and @c1728p9) to work out the solution together that works best for mbed? I don't have an EWARM7 testbench, here. A PR with a fix is very appreciated. 

Thanks,
Jonatan

c1728p9, 2018-09-25:
Closing this issue since #425 has been merged.

closed 2018-09-25 by c1728p9