ARM-software/CMSIS_5 #419: Do not manually unroll loops

smeijer1234, 2018-08-28:
Many of the DSP kernels have loops manually unrolled; most of the times they are manually unrolled 4 times.

There are several problems with this approach:
- Manual unrolling does not give very portable or future proof performance: unrolling by 4 might work for one core, but not very well for another.
- Manually unrolling loop is interfering with compiler analysis and transformations. Thus, this may easily confuse compilers and may lead to suboptimal decisions and performance.
- Code bloat in the library; this is #defined, so not a problem for the final compiled image, but the implementation of the library.

A good first example is the simple addition of 2 vectors:

https://github.com/ARM-software/CMSIS_5/blob/d143b79da1e2734684fe29a987a1eeae9e2c8882/CMSIS/DSP/Source/BasicMathFunctions/arm_add_f32.c#L41

As described in the comments, this is not doing more than:

```
for (n = 0; n < blockSize; n++)
  Dst[n] = SrcA[n] + SrcB[n];
```
which is almost literally how we should implement this. To specify unrolling, we could add pragmas:

```
#pragma unroll 4
for (n = 0; n < blockSize; n++)
  Dst[n] = SrcA[n] + SrcB[n];
```
which is not only conciser and resulting in the same performance results, it also allows easy experimentation with the unroll factor.

llefaucheur, 2018-09-03:
Hi smeijer1234, one other option is to port the code in assembly and optimize the instructions based on processor's specific characteristics, like dual-issue capabilities of CM7. This last option guarantees the best performance for your application, while the generic code is giving a good teaching starting point. Regards, Laurent

christophe0606, 2019-05-29:
We now have a global compilation flag to switch between unrolled or not unrolled version. We have to support several compilers and for some of them writing unrolled loops is the right way.

closed 2019-05-29 by christophe0606

madcowswe, 2019-05-31:
Have you tried adding the `restrict` keyword? It is specifically to help the compiler add optimizations, and sometimes even vector instructions.