ARM-software/CMSIS_5 #422: Tiny Matrix Dimensions

smeijer1234, 2018-08-28:
The matrix dimensions used in the Matrix tests are so absolutely tiny with sizes like1x1, 1x2, 2x4, that it is probably not even worth calling a generic matrix multiply routine for this:

https://github.com/ARM-software/CMSIS_5/blob/d143b79da1e2734684fe29a987a1eeae9e2c8882/CMSIS/DSP/DSP_Lib_TestSuite/Common/src/matrix_tests/matrix_test_common_data.c#L163

There are several problems with these small matrix dimensions:
- if these matrix dimension are representative for this domain, then we probably want more specialised helper routines for this (we don't need unrolled loops, remainder loops etc.),
- these sizes are so small that the manually unrolled loop bodies are executed at most 1 time since they are unrolled 4 times. Thus, if the argument to use these small sizes is that this is enough for testing, that this is testing the edge cases, I am not so sure since I don't think the unrolled loop body is executed twice (or more).
- but also importantly, if these tests are also used for performance measurements, it's easy to fall in this "trap" because the clock cycles are reported after all, we are not measuring the hot-loop, because there is no hot loop. I.e., more time is spent in other code that we are actually not interested in, or should not really be interested in (but correct me if I am wrong).

I would suggest we also add some tests that feed the kernels with some data, so that they can do a little bit of number crunching and that the reported cycle count actually have a meaning. Currently that is not the case, and if adding tests, and increasing matrix domains is somehow not desired, we should at least consider removing some of the meaningless tests (e.g. matrices of 1x1), and replace them with bigger dimensions that tests the same code paths, or stop reporting cycle counts for these tests.

llefaucheur, 2018-09-03:
Hi smeij1234 you are true. One other option is to implement something like [this](https://developer.apple.com/documentation/accelerate/simd?language=objc) and have dedicated APIs for lengths below 4. Regards, Laurent.

christophe0606, 2019-05-29:
We have plans to rework the test framework and add more meaningful test patterns.

closed 2019-05-29 by christophe0606